cjcliffe/CubicSDR #284: Rig Control always enabled?

Toontje, 2016-02-14:
When i enable rig control AND when i disable rig control using the menu i get this message in the log:
```
Rig thread starting.
mir_sdr_SetRf: f->136214760.000Hz (int=16 frac=83b afc=614) fSynth:2179436160.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=14166760.0+0.0+2048000.0Hz+120000000.0Hz=136214760.0Hz RfToggle->0
mir_sdr_ReadPacket: Rf update confirmed: Rf=136214760.0Hz RfToggle=0
mir_sdr_ResetUpdateFlags: resetGainUpdate=0 resetRfUpdate=1 resetFsUpdate=0
```

Does that mean that the Enable/Disable toggle doesn't work and rig control is always enabled?

cjcliffe, 2016-02-14:
It should terminate the rig control when disabled but it's possible I've broken something with recent menu changes; will double-check

cjcliffe, 2016-02-15:
There were some problems with how the rig was being initialized; and if it failed the menu didn't reflect that -- now it should handle hamlib failures/disconnects and un-check the menu to indicate that it stopped/didn't start.

closed 2016-02-15 by cjcliffe