cjcliffe/CubicSDR #289: Consider adding Jack support to the build to allow easier virtual audio support

Analias, 2016-02-15:
On Mac OS X, I am running Jack as an alternative to creating virtual audio ports in OS X at the system level.  This is working, but the initial configuration is somewhat cumbersome and order dependent. It is easy to do out of order and have to restart my system to recover.  Allowing Jack to be enabled at build time and linked in should simplify some of that setup and make it easier to configure and use virtual audio ports with other applications (fldigi, WSJT-X, etc.).  Supporting Jack would be applicable for use with Linux and Mac OS X.

k9wkj, 2016-02-16:
as troublesome as jack can be
it would real nice to be able to route things on purpose
and not by accident!

Analias, 2016-02-16:
I noticed that the current code base includes code to use Jack as part of RTAudio.  I'll have to look closer to see how it is used.

Analias, 2016-02-16:
Well, it's not as simple as copying the config in the CMakeList.txt to include with the mac build.  Time to go look closer again.

cjcliffe, 2016-02-16:
@Analias I've only used the JACK audio support on Linux so far and it seemed to work well; I haven't tried it yet on OSX (been using Soundflower) but I'll give it a go and see where the issue lies.

Toontje, 2016-02-16:
Yeah, Jack is probably more flexible than Soundflower, but Soundflower is
so much easier to use.