cjcliffe/CubicSDR #290: The audio panel should apply to the currently selected modem only

Analias, 2016-02-16:
It appears that the audio panel is an aggregate of all the audio signals generated by the modems.  I would expect that the audio panel would apply only to the currently selected modem.  This would be consistent with the implemented behavior where each modem has it's own audio sink (which works quite nicely thank you - makes routing to virtual audio ports and external applications easier).

cjcliffe, 2016-02-16:
@Analias the "Audio Output" only applies to the current modem so you should already be able to route each modem to a separate device if desired; the "Audio Sample Rate" is shared between devices on the same output since CubicSDR has an internal mixer that can balance all the signals based on relative gain and doesn't open multiple instances of the audio device.

closed 2016-03-22 by cjcliffe

neilwillgettoit, 2016-05-15:
How does this work with pulseaudio in linux ? The only audio output option I get is a single 'pulseaudio' option even though I've got multiple quite a few options within pulse to route to different applications. Currently I've been using pavucontrol to route audio around, but it would be nice to send different modems to different pulseaudio options.