cjcliffe/CubicSDR #292: Tuning gets screwy at times

Analias, 2016-02-19:
I'm not sure what is up, but at times the tuning doesn't follow the frequencies shown in tuning bar and seem to be way off according to the bottom FFT/Waterfall.  I've have included a picture to show what I mean.  Noticed the modem in the bottom FFT/waterfall says 5.551MHz, but is clearly off by almost 70KHz.  The modem doesn't even show in the modem FFT/Waterfall even though the range includes the frequency listed in the tuning bar.  

This was after tuning around by clicking on the digits in the tuning bar and using the keyboard to move.

-Freeman

[image]

Analias, 2016-02-19:
I should also say that I'm hearing the audio that is graphically shown in the bottom waterfall.

cjcliffe, 2016-02-19:
@Analias is this only related to the new delta-locking feature or are you seeing it when unlocked too?  It's possible something's getting mis-aligned since I added some some extra visual checks for delta-lock modems to keep them from jumping around visually when changing center frequency.

Modems are spawned fresh each time and old ones are garbage collected.  I'd say it's just a label update problem but the frequency bar clearly shows that same frequency but at the same time it's not actively selected so it's possible that the frequency shown isn't actually attached to that modem.

Analias, 2016-02-19:
@cjcliffe I’ve tried several times to reproduce the problem.  I tried with both the delta lock on and off.  It’s not easy to cause, which is one reason why I took a screen shot to show the problem and prove to you and me I’m not insane <grin>

I’ll keep banging on it and see if I can way to reliably reproduce the behavior.

-Freeman, N5FPP

cjcliffe, 2016-03-08:
I'll close this one up for now, if you find a solid way to reproduce it please open up a new issue; doesn't seem too serious either way.

Cheers!

closed 2016-03-08 by cjcliffe