cjcliffe/CubicSDR #293: Build errors on Mac

sean808080, 2016-02-20:
So I'm a newbie at building projects like these.  I'm probably screwing something up..

First, following the wiki build instructions leads to an error.  The cake step in CubicSDR fails with this:
CMake Error at cmake/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find wxWidgets (missing: wxWidgets_FOUND)
Call Stack (most recent call first):
  cmake/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindwxWidgets.cmake:869 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:136 (find_package)

The only way to get cmake to work from the command line is by adding the wxrc_executable parameter like this:
cmake ../ -DwxWidgets_wxrc_EXECUTABLE=/Users/sean/Dev/wxWidgets-staticlib/bin/wxrc  -DwxWidgets_CONFIG_EXECUTABLE=/Users/sean/Dev/wxWidgets-staticlib/bin/wx-config  -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1

Now I'm stuck at the cpack step:

CPack: - Install project: CubicSDR
error: install_name_tool: can't open input file: /Users/sean/Dev/CubicSDR/cmake/x64/CubicSDR.app/Contents/MacOS/libltdl.7.dylib for writing (Permission denied)
error: install_name_tool: can't lseek to offset: 0 in file: /Users/sean/Dev/CubicSDR/cmake/x64/CubicSDR.app/Contents/MacOS/libltdl.7.dylib for writing (Bad file descriptor)
error: install_name_tool: can't write new headers in file: /Users/sean/Dev/CubicSDR/cmake/x64/CubicSDR.app/Contents/MacOS/libltdl.7.dylib (Bad file descriptor)
error: install_name_tool: can't close written on input file: /Users/sean/Dev/CubicSDR/cmake/x64/CubicSDR.app/Contents/MacOS/libltdl.7.dylib (Bad file descriptor)
CMake Error at /opt/local/share/cmake-3.4/Modules/BundleUtilities.cmake:787 (message):
  Command failed:

   'install_name_tool' '-id' '@executable_path/../MacOS/libltdl.7.dylib' '/Users/sean/Dev/CubicSDR/cmake/x64/CubicSDR.app/Contents/MacOS/libltdl.7.dylib'
Call Stack (most recent call first):
  /opt/local/share/cmake-3.4/Modules/BundleUtilities.cmake:848 (fixup_bundle_item)
  /Users/sean/Dev/CubicSDR/cmake/cmake_install.cmake:34 (fixup_bundle)

CPack Error: Error when generating package: CubicSDR
MacBookAir:cmake sean$ 

All I want is the latest version!  Lol

cjcliffe, 2016-02-21:
@sean808080 looks like you may have some cmake build/target files locked as root by doing a 'sudo cpack' (which shouldn't be required).

You should be able to re-set your build folder with:
```
sudo chown -R sean:sean ~/Dev/CubicSDR
```

Additionally it looks like you might have generated your build files in the CubicSDR/cmake source folder  '/Users/sean/Dev/CubicSDR/**cmake**/x64/CubicSDR.app/' -- the included CubicSDR/cmake folder is actually a resource folder which contains scripts (such as one for finding wxWidgets) as well as some important resources that are used to generate build files -- chances are building there will corrupt the build.

For wxWidgets, the only thing you should need to actually set is the -DwxWidgets_CONFIG_EXECUTABLE=.. parameter; the rest is deduced from using wx-config executable.

I'd recommend checking out CubicSDR again and make sure to build it in a folder you've created yourself such as ~/Dev/CubicSDR/build.

sean808080, 2016-02-21:
Wow...thanks so much for pointing that out.  All I had to do was delete the CubicSDR folder and start again.  I got confused by the wiki step that said:

ccliffe$ cd CubicSDR/cmake_build

As I didn't have that directory, I thought it was supposed to be cmake LOL

Suggest adding the mkdr so it's clear that it's a new folder.

Still seeing errors but there is a binary that runs.  I think it'l work....guess we'll find out!

Thanks and 73

cjcliffe, 2016-02-21:
@sean808080 good point; I've updated the OSX build wiki for path clarity -- cheers!

closed 2016-02-23 by cjcliffe