cjcliffe/CubicSDR #294: window snapshot

k9wkj, 2016-02-21:
how about a quickie screenshot that just grabs the window contents
and saves the contents as a png
im always looking for errant transmitters and this would be much handier
than using a separate tool

i know....  i know...  im contributing to feature bloat...

Analias, 2016-02-21:
Personally, I'd like to see this if it is for the currently focused modem display or bandwidth display only.  All the currently supported OSes support screen shots from the OS level.

k9wkj, 2016-02-22:
well a screen shot we dont want
just a window capture
without having to use a extra tool
just a shortcut  to snap the window and doesnt one need to know filter
width and freq?

cjcliffe, 2016-02-23:
@k9wkj I'm hoping to be able to spool out the waterfall to .PNG files of desired resolution; it's on the https://github.com/cjcliffe/CubicSDR/wiki/CubicSDR-Roadmap-and-Ideas list along with future possibility of rendering to video files and a few related ideas.

closed 2016-08-16 by cjcliffe