cjcliffe/CubicSDR #295: Waterfall inconsistent...

Analias, 2016-02-22:
Could you explain this please - I'm confused by what is being shown between the modem and bandpass waterfalls.  In the attached screenshot, you see a strong signal under 18.10Mhz in the modem display.  This same signal looks identical to the signal under 18.13Mhz.  Is the signal offset in the wrong direction?

What confuses me even more, when a signal appears in the modem bandpass it appears the same in both displays.

At times some signals show in one display but not the other, and vice versa.  A signal can be in either display and show no trace of it in the other.  

I don't know if this is related to the behavior in listed in the other tracker about tuning.

Analias, 2016-02-22:
Oddly, when I see a sweeper signal get detected its movement in the displays is consistent with the direction it is changing.

cjcliffe, 2016-02-23:
@Analias First note is that CubicSDR can introduce aliasing at some zoom levels since it uses a combination of decimation and visual FFT resampling instead of just scaling a massive FFT.

The primary waterfall display data is sourced from the SDR input where as the modem waterfall is sourced from the polyphase filterbank channel the modem is occupying.  Also main waterfall uses a visual DC-spike hiding trick and modem waterfall will perform actual DC-correction (it's sourced from the actual data going into the modem).

For the channelizer the 'Channel Count' = 'InputRate / 500kHz' -- with the first channel centered at the center frequency.  Channel bandwidth can vary a bit depending on the input sample rate and it's divisibility by 500kHz.

Additionally the filterbank channelizer will wrap a certain percentage of the signal in each 500kHz channel from edge to edge -- so if your signal is on the left edge of the center channel and another is on the right edge it will appear to the left of the left signal and vice-versa in modem view.

So if you change the center frequency you should be able to eliminate aliases from the modem view and zooming in on the main waterfall can help identify/suppress aliases as well thanks to the decimation.

Although it's a bit strange, it is expected behaviour due to the way the components work -- I may be able to make the channelizer alignment behave better (adapt alignment/size to active modems) or make the channel size configurable in the future.  

I feel like adding an option to let you visualize the channel boundaries might be useful to explain and visualize what's happening here -- I'll have to think about that.

cjcliffe, 2016-02-24:
@Analias for a visual representation of that gibberish here's what I'm guessing the polyphase filterbank channelizer bands are in your screenshot:

[image]

So the channel that's feeding the modem looks like this:

[image]

In order for the waterfall to center the frequency (and the modem to demodulate the signal) it needs to mix the signal up so that your modem is at the baseband, in doing so there's only a limited amount of bandwidth per channel, so what happens is the right side of the channel is rotated to the left side, and you're left with the resulting signal mix:

[image]

Which looks remarkably like your modem waterfall :)

Hope that helps!

closed 2016-03-08 by cjcliffe

Toontje, 2017-11-25:
What about getting rid of the modem waterfall and just improve the main waterfall pan and zoom functions? Or am i missing the point of having a modem waterfall?