cjcliffe/CubicSDR #298: Multiple demodulators

vsboost, 2016-02-23:
Hi Charles,

Not sure if this is normal operation or not but i noticed that if you drag the waterfall to a position where the current demodulator is no longer present and click on a part of the waterfall it creates a new one, similar to Shift Click.

Running 0.1.24-alpha on Win8.1 with a RTL dongle

The below screenshot created two demodulators without holding shift.

[image]

cjcliffe, 2016-02-23:
@vsboost it's normal operation for CubicSDR to create a new modem when there's not currently an active one to move (active = center frequency within bandwidth) and there's no limit on how many modems you can leave lying around in a session :)  You can also delete the current modem and if you click again without selecting/hovering another it'll create a new one.

The gain value, mode, bandwidth and squelch, etc. should be carried from the last active modem; I'll check to ensure that the gain is actually being updated.

vsboost, 2016-02-23:
Ahh cool,

Ill leave this open until you check the audio levels.

george-viaud, 2016-02-25:
Looking into this, it appears that gain is copied from the previously active demodulator unless it's off the waterfall in which case it returns to default value.

Is the desired / described behaviour as follows(?) :

1) If no demodulator has ever been selected during this session, use default value.
2) If a demodulator has been previously selected, use the last value selected even if off waterfall

If so I'll gladly work on this.  Elsewise please help me understand the need and I'll try to fill it.

Thanks!

vsboost, 2016-02-25:
Correct, normally i like to scroll through the waterfall and click on  signals of interest and find the audio gain levels default, would be great if they take the gain value from previous demodulator even if they are off screen.

Cheers

cjcliffe, 2016-02-26:
@george-viaud this https://github.com/cjcliffe/CubicSDR/blob/master/src/demod/DemodulatorMgr.cpp#L279 should be being used to manage the last used gain level; it's possible it's not being used for new modems but that was the idea behind those 'last*' memory functions.

cjcliffe, 2016-03-08:
Found the bug here and committed a patch for it; looks like I had tried to fix it earlier -- but didn't think it through and just had the last gain update in the wrong place.

closed 2016-03-08 by cjcliffe