cjcliffe/CubicSDR #299: Feature request: station lookup of currently demodulated broadcast signal

dc1rdb, 2016-02-23:
As implemented in the latest extio for SDRPlay, would it be possible to extend CubicSDR to use the world-wide station list maintained by http://www.eibispace.de/ to display the name of the likely stations upon selection of a particular broadcast signal?

SDRplay, 2016-02-23:
As we are in the process of redoing the SoapySDRPlay we will be talking to Charles about this and other features. Our intention is to bring SoapySDRPlay up to the same level as the ExtIO interface.

cjcliffe, 2016-02-24:
@dc1rdb this sounds like an excellent use case for a LUA module so I'll add it to the enhancement list.

@SDRplay looking forward to it and getting started on #194 when we have the chance to hash it out after 0.2.0-beta release.