cjcliffe/CubicSDR #302: Feature Request:  Squelch-Break indicator

mfgeo, 2016-02-25:
When monitoring multiple signals in a band, it's sometimes hard to determine which 'modem' is breaking squelch.

It would be useful to have some form of indicator to display which has broken squelch (sometimes several at the same time).  Perhaps the bandwidth-bars could change color for example, or the indicated frequency above the bars, etc?

cjcliffe, 2016-02-25:
I'll think about this; in Solo mode the squelch-break is obvious as it's highlighted in yellow but I think briefly coloring the modem and then fading out would be a reasonable indicator in non-solo mode.

cjcliffe, 2016-06-04:
The waterfall will now flash the demodulator (with the current hover theme color) for about a second and fade out when squelch is broken -- this also opens the door for other visual cues; thanks for the idea 👍 

Cheers!

closed 2016-06-04 by cjcliffe