cjcliffe/CubicSDR #304: Feature Request:  Auto-Squelch

mfgeo, 2016-02-25:
I'm not sure if the squelch scale is linear, logarithmic, exponential, etc but I find that (for very weak signals) it is very difficult to set the squelch so that it breaks only when signal is present.

Looking in the session files, it appears that squelch can be rather finely controlled.

It would be useful if there was a way to (by click of a button or similar) set the squelch just above the noise floor at that moment by some margin (perhaps configurable or calculated in a smart way using standard deviation over a window of samples?)

This would let the user quickly floor the squelch to a safe, quiet level that is likely to trigger on key (or, at least that's the user story =) )

Thank you

george-viaud, 2016-02-25:
Ahh I see, looks like squelch is in dB:
currentSignalLevel = linearToDb(accum / float(inp->data.size()));

I've since added this feature and a mouse-wheel fine-adjust for squelch to pull-request 305.

In my opinion it works really well but I still have some issues with very small signals (they fail to trigger the squelch break).  I'll do more work on that and add a new ticket as needed.  Thanks

cjcliffe, 2016-02-25:
@george-viaud thanks for the PR -- feel free to tinker some more with the squelch logic; it's a feature I really haven't put too much additional thought into so far.  

I'm thinking the squelch ceiling and floor should adapt to the visual ceiling/floor of the spectrum view (in dB) so that it's at least tied to something relative to the signal instead of fixed values.

closed 2016-04-02 by cjcliffe