cjcliffe/CubicSDR #308: Hamlib support for using CubicSDR as controllable receiver

Analias, 2016-03-01:
I don't think this is a duplicate, if it is please delete it.

The current rig control is great if you want to have a second radio slaved to CubicSDR.  Adding support for CubicSDR to act as a controllable radio from hamlib clients would be very useful also. A very useful example of how this can be done is included as part of gqrx. Doing so would allow applications like WSPR-X or gpredict to control CubicSDR directly.

I would recommend implementing the Hamlib Net RIGCTL (id = 2) protocol. 

-Freeman

Marzona, 2016-05-15:
Hello,

I currently work on my spare time on rig-remote (https://github.com/Marzona/rig-remote) and I would appreciate if CubicSDR exposes a tcp/ip socket for sending CubicSDR rigctl commands from rig-remote.
TCP/IP would allow to have the features of rig-remote to be used with CubicSDR too with no effort on CubicSDR developers.

neilwillgettoit, 2016-05-16:
I think you can do exactly what you're looking to do already in the master branch up to date version. For example I have wsjtx controlling cubicsdr over tcp by: 

1. Start rigctld and bind 12 7.0.0.1:7076
2. Start CubicSDR latest git clone build with hamlib support eenabled -
3. Rig Control > Com / Port > Set to 127.0.0.1:7076 
4. Rig Control > Select Control Rig, Follow Rig, Track Modem
5. Rig Control > Model > Hamlib NET rigctl 
6. Rig Control > Enable Rig 
7. Control cubicSDR from rigctld or application interfacing with rigctld

A-j-K, 2022-08-04:
Still can't do this on 0.2.7 so I guess the milestone got missed?