cjcliffe/CubicSDR #309: 1.25 Windows binaries bug?

jdeitch, 2016-03-10:
@cjcliffe 
In the latest Windows binaries the mouse wheel no longer changes frequency.  Previous versions and linux 1.25 are ok.

Win7 x64 SP1

Jim

jdeitch, 2016-03-10:
> Have you tried to reboot your machine, I had something similar.

Many times.

cjcliffe, 2016-03-10:
@jdeitch might be broken due to the new global keyboard handler; it's possible I disconnected some events by mistake -- I'll take a look.

cjcliffe, 2016-04-02:
@jdeitch can't reproduce here in Win 10 with https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.27-alpha windows build -- hopefully this is fixed up now.

jdeitch, 2016-04-06:
@cjcliffe  That build still has the problem of scroll wheel not changing frequency.  Win 7 Pro x64 SP1
Also, when saving a session the frequency and the center frequency are saved on different lines, but the center frequency is restored to both.

cjcliffe, 2016-04-06:
@jdeitch session load issue sounds like a possible typo or bug -- I'll take a look at that.  The scroll wheel is working here in Win10; does the main waterfall zoom and squelch level work with scroll wheel or is it broken in general?

cjcliffe, 2016-04-06:
@jdeitch I can't seem to reproduce the session loading issue; and I've reviewed the code -- it shouldn't be able to do that from what I can tell.   Can you post one of your session files?  Perhaps there's something different going on.

jdeitch, 2016-04-07:
@cjcliffe Scroll wheel not working at all.  Here is a session file I just made for this post.  Audio gain and squelch level load correctly, demodulator and bandwidth load correctly, center frequency loads correctly.  Tuned frequency and gain settings do not load correctly.

```
<?xml version="1.0" ?>
<cubicsdr_session>
    <header>
        <version>0.1.27-alpha</version>
        <center_freq>118013482</center_freq>
        <sample_rate>2048000</sample_rate>
    </header>
    <demodulators>
        <demodulator>
            <bandwidth>6000</bandwidth>
            <frequency>118574360</frequency>
            <type>AM</type>
            <squelch_level>-29.7802</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Default Device</output_device>
            <gain>1</gain>
            <muted>0</muted>
            <active>1</active>
        </demodulator>
    </demodulators>
</cubicsdr_session>
```

vsonnier, 2016-04-07:
I'm exactly in the same situation, running 0.1.27 Win64 binary on Windows 7. Scroll is not working. Loading session works if loaded from startup, then (not always alas, but very often) if I load another, the frequencies are not refreshed and the display do not align itself on the new demodulators. Other times, re-loading session duplicates some demodulators on-themselves.... 
The "Reset session" and then loading a session file once again as no effect and the frequencies are not restored, the only way is to restart the application.
This looks like a defect in complete reset/cleanup/refresh of the current session properties...
Alas, I've yet to find an easy reproducible test case like @jdeitch found for its particular problem.

BTW @cjcliffe, amazing program I use it instead of SDR#, and donated to support you. I especially like the adaptative color dynamic range, that allows to see with maximum contrast everytime. I think SDR# don't do this. Thank you!

jdeitch, 2016-04-07:
@cjcliffe FYI..I tried both the 64bit and 32bit builds with the same results.

cjcliffe, 2016-04-07:
@vsonnier thanks for the confirmation; I'm going to see if I can try it here on Windows 7 -- that seems to be the common element here.

@jdeitch your session file looks fine; there must be something else going on here.  

The mouse issue could be related to the wxWidgets change but it's possible some of the warnings I'm overlooking in MSVC could be a problem; this sounds like possible memory corruption to me (demodulators values loading improperly) -- there's also an update for Visual Studio which I have not yet applied.

If you're able try installing older versions until you find one that doesn't have the issue -- this might help me pinpoint relevant changes.

Another experiment you might try with 0.1.27 is create a few demodulators after a session load and then use the 'Tab' key to switch between them; re-order them and confirm that the tabbing order matches the new on-screen order.  I've found in the past when the demodulator manager stack is corrupt this sorting is usually the first thing to break; they'll tab out-of-order or skip completely -- a full clean and recompile fixed it in those cases.

cjcliffe, 2016-04-08:
@vsonnier thanks for your support and I'm glad it's working well for you! I've always been bothered by having to constantly fidget with the range sliders, buttons and other bits in several applications so I've tried my best to create an interface where those things aren't really necessary.

jdeitch, 2016-04-08:
@cjcliffe 1.25 was the last version the scroll wheel worked under Windows 7 x64.  Trying your demodulator test now.

cjcliffe, 2016-04-08:
@jdeitch I'm spinning up a Windows 7 VM so I should be able to get a dev environment in there soon to debug it.   The main change since 1.25 would likely be wxWidgets 3.1.0 which I'm guessing may have a bug in Windows 7.   I may end up making a few patches for wxWidgets for this and their new Linux OpenGL init if a pull from wx master doesn't fix it :)

vsonnier, 2016-05-03:
@cjcliffe I have a (somewhat reproducible) use case for duplicated, stack upon themselves demodulators showing up at loading of an XML session file:

- launch CubicSDR,
- choose RTL-SDR,
- uncheck auto gain, set 25dB instead,
- load DMR_430Mhz_ref.xml:

`````xml
<?xml version="1.0" ?>
<cubicsdr_session>
    <header>
        <version>0.1.27-alpha</version>
        <center_freq>430263415</center_freq>
        <sample_rate>2560000</sample_rate>
    </header>
    <demodulators>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430575946</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.75</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430287741</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>11660</bandwidth>
            <frequency>430263415</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>0</muted>
            <active>1</active>
        </demodulator>
        <demodulator>
            <bandwidth>15000</bandwidth>
            <frequency>430100589</frequency>
            <type>FM</type>
            <squelch_level>-29.7744</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Default Device</output_device>
            <gain>0.125</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430050956</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430175376</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
    </demodulators>
</cubicsdr_session>
````

Now between 1/2 and 1/3 times, a demodulator get reloaded with duplication over itsef, here the 430.263Mhz. Also the Delta Lock (V) looks activated for this demodulator:

![dmr_430mhz_duplicated_demodulator](https://cloud.githubusercontent.com/assets/5152742/14992107/b2568d8e-1164-11e6-8320-212a2e10eca5.jpg)

- If you reload the XML file, the demodulators looks OK:
![dmr_430mhz_reload_ok](https://cloud.githubusercontent.com/assets/5152742/14992159/f9902c6e-1164-11e6-862a-bb3b7e1a4392.jpg)
 
Here is the global config.xml file of CubicSDR:
``````xml
<?xml version="1.0" ?>
<cubicsdr_config>
    <window>
        <x>-9</x>
        <y>-9</y>
        <w>1920</w>
        <h>968</h>
        <max>1</max>
        <tips>0</tips>
        <low_perf_mode>0</low_perf_mode>
        <theme>5</theme>
        <snap>1</snap>
        <center_freq>430263415</center_freq>
        <waterfall_lps>51</waterfall_lps>
        <spectrum_avg>0.549296</spectrum_avg>
    </window>
    <devices>
        <device>
            <id>CABLE Output (VB-Audio Virtual Cable)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Default Device</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Generic RTL2832U OEM :: 00000001</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts>
                <buffers>15</buffers>
                <bufflen>16384</bufflen>
            </streamOpts>
            <settings>
                <direct_samp>0</direct_samp>
                <iq_swap>false</iq_swap>
                <offset_tune>false</offset_tune>
            </settings>
            <rig_ifs />
        </device>
        <device>
            <id>Microphone (Realtek High Definition Audio)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Mixage stéréo (Realtek High Definition Audio)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
    </devices>
</cubicsdr_config>
``````

Also very often some time afterwards (< 5mn) CubicSDR crashes, but I don't know if its related of not.

Maybe you can provide binaries with Core Dump (minidump) functionality. If I remember correctly, you can install a handle of unhandled exeptions and trigger a minidump by loading for instance MiniDumpWriteDump entry point from DBGHELP.dll for instance. 
Maybe there are more direct ways to do so nowaydays. Anyway that could be very useful to debug crashes, provided you compile binaries with debug info enabled.

Sorry for the spam:)

cjcliffe, 2016-05-03:
@vsonnier thanks! that's exactly the type of info and test case I needed; hopefully I can reproduce with the same steps provided.

vsonnier, 2016-05-04:
@cjcliffe I just noticed it does the same if you replace 
``````xml
<output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
``````
by
``````xml
<output_device>Default Device</output_device>
````````
everywhere in the session file, so you may not need to have Virtual Audio Cable installed to reproduce the bug.

jdeitch, 2016-05-16:
@cjcliffe Just did a complete new, clean install of both windows builds of 1.28-alpha and the reported bugs still exist.

cjcliffe, 2016-05-17:
@jdeitch aye, sorry I haven't had a chance to rebuild it in Windows 7 yet to locate the issue but It's on the short list :)

cjcliffe, 2016-05-27:
Ok, after building a few windows 7 vms and churning back through some commits I found that during the global key function updates I removed a piece of code that didn't do what I thought it did.

```
	if (demodTuner->getMouseTracker()->mouseInView()) {
		if (!demodTuner->HasFocus()) {
			demodTuner->SetFocus();
			
		}
	}
	else if (!wxGetApp().isDeviceSelectorOpen() && (!modemProps || !modemProps->isMouseInView())) {
		if (!waterfallCanvas->HasFocus()) {
			waterfallCanvas->SetFocus();	
		}
	}
```

I wrongly assumed this was to keep keyboard focus on the screen elements; but in reality it was from earlier work in Windows 7 when it was necessary to focus the window to make it accept the scroll wheel event..  

I'll have a patch in soon and I'll post a test binary here shortly.

cjcliffe, 2016-05-27:
These builds should fix the windows scroll wheel issue:

[CubicSDR-0.1.28-win64.exe.zip](https://github.com/cjcliffe/CubicSDR/files/285718/CubicSDR-0.1.28-win64.exe.zip)

[CubicSDR-0.1.28-win32.exe.zip](https://github.com/cjcliffe/CubicSDR/files/285719/CubicSDR-0.1.28-win32.exe.zip)

Going to reboot and try them in the Win7 VM here.

vsonnier, 2016-05-27:
@cjcliffe Thanks for your builds. Now what I can see on controls :
-  Freq / Bandwith / Center : OK
-  Sound / Peak level bars : OK
- Waterfall speed bar: Works, but changes are incredibly slow
- Squelch level bar: idem above
- GnzrWaterfall

cjcliffe, 2016-05-27:
@vsonnier thanks for the feedback; I actually hadn't realized that the code that was contributed for the squelch meter was actually working for the other meters now too..  I'll have to add the additional scale factor for the various ranges they use to make it more usable; I'll also add support to the gain sliders.

The vibrating top left demodulator view is normal, it keeps re-adapting to the demodulator range; I'll likely add some code to cancel the mouse wheel there.

I'm going to open a new issue for that and the session issue shortly.

cjcliffe, 2016-05-27:
I've split the remaining issues into #353 and #354 -- closing this one up for now; thanks!

jdeitch, 2016-05-28:
@cjcliffe   Just tried the above new binaries and ran across a weird problem.  Uninstalled all previous version just to be safe.  Installed and started the new x64 program.  everything looks great.  Placed the mouse on the bandwidth frequency and wheeled down.  It worked!  Then I clicked on a random spot on the waterfall to choose a frequency and nothing.  Hmm..that is weird.  clicked again.  Still nothing.  I realized I hadn't chose a device.  Saw the tab for choosing a device on the taskbar, clicked on it and viola, I can choose a device.  Well, not really.  If I mouse over the main CubicSDR screen it takes focus and the choose device screen is hidden.  I have to shrink the main screen to choose a device.  X32 version acts the same.  After that everything works great just like vsonnier says.

Jim

cjcliffe, 2016-05-28:
@jdeitch ah, yes I should make sure that the focus isn't stolen while the device dialog is open -- I'll look into that.

jdeitch, 2016-05-28:
@cjcliffe  Thank you for affirming that I was not losing my mind!

vsonnier, 2016-05-28:
I indeed observe the same problem as @jdeitch :)

cjcliffe, 2016-05-31:
Focus issues now fixed over in #353

closed 2016-05-31 by cjcliffe