cjcliffe/CubicSDR #310: Random crash on OS X

EvanKrall, 2016-03-11:
I was playing around with CubicSDR on my Macbook Air / HackRF One, looking at the broadcast FM bands; after a few minutes, it crashed. Here's the crash report:

```
Process:               CubicSDR [88533]
Path:                  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:            com.cubicproductions.cubicsdr
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           CubicSDR [88533]
User ID:               501

Date/Time:             2016-03-10 23:26:09.080 -0800
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        C59B7757-2928-934D-4C2A-DBFABEDFFC3B

Sleep/Wake UUID:       93BF3B7D-3FDD-4F6F-AEE7-5EB6ED82B808

Time Awake Since Boot: 550000 seconds
Time Since Wake:       1200 seconds

System Integrity Protection: disabled

Crashed Thread:        14

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x18:
--> 
    __TEXT                 0000000101f9c000-0000000102593000 [ 6108K] r-x/rwx SM=COW  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_pthread.dylib       	0x00007fff8fab8434 pthread_mutex_lock + 0
1   com.apple.CoreFoundation      	0x00007fff8d1bbd5f __CFRunLoopDoObservers + 319
2   com.apple.CoreFoundation      	0x00007fff8d1ad908 __CFRunLoopRun + 872
3   com.apple.CoreFoundation      	0x00007fff8d1ad338 CFRunLoopRunSpecific + 296
4   com.apple.HIToolbox           	0x00007fff90195935 RunCurrentEventLoopInMode + 235
5   com.apple.HIToolbox           	0x00007fff90195677 ReceiveNextEventCommon + 184
6   com.apple.HIToolbox           	0x00007fff901955af _BlockUntilNextEventMatchingListInModeWithFilter + 71
7   com.apple.AppKit              	0x00007fff886810ee _DPSNextEvent + 1067
8   com.apple.AppKit              	0x00007fff88a4d943 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
9   com.apple.AppKit              	0x00007fff88676fc8 -[NSApplication run] + 682
10  com.cubicproductions.cubicsdr 	0x00000001022366b6 wxGUIEventLoop::OSXDoRun() + 166
11  com.cubicproductions.cubicsdr 	0x000000010214e517 wxCFEventLoop::DoRun() + 39
12  com.cubicproductions.cubicsdr 	0x000000010209c6eb wxEventLoopBase::Run() + 235
13  com.cubicproductions.cubicsdr 	0x0000000102058cd2 wxAppConsoleBase::MainLoop() + 194
14  com.cubicproductions.cubicsdr 	0x0000000102058869 wxAppConsoleBase::OnRun() + 25
15  com.cubicproductions.cubicsdr 	0x000000010226c55e wxAppBase::OnRun() + 62
16  com.cubicproductions.cubicsdr 	0x00000001021e2c95 wxApp::OnRun() + 37
17  com.cubicproductions.cubicsdr 	0x00000001020d3dda wxEntry(int&, wchar_t**) + 298
18  com.cubicproductions.cubicsdr 	0x00000001020d3f3f wxEntry(int&, char**) + 63
19  com.cubicproductions.cubicsdr 	0x0000000101fa3510 main + 48
20  libdyld.dylib                 	0x00007fff8f9635ad start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8c0fbff6 kevent_qos + 10
1   libdispatch.dylib             	0x00007fff8b1f9099 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff8b1f8d01 _dispatch_mgr_thread + 52

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff8c0f5386 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8c0f47c7 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff8d1ae624 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff8d1adaec __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff8d1ad338 CFRunLoopRunSpecific + 296
5   com.apple.AppKit              	0x00007fff88740065 _NSEventThread + 149
6   libsystem_pthread.dylib       	0x00007fff8fabac13 _pthread_body + 131
[... 521 lines cut ...]
__IMAGE                            528K        2 
__LINKEDIT                       106.0M       46 
__TEXT                           186.3M      268 
__UNICODE                          552K        2 
mapped file                       76.9M       18 
shared memory                     16.4M       11 
===========                     =======  ======= 
TOTAL                            695.4M      820 
TOTAL, minus reserved VM space   692.6M      820 

Model: MacBookAir6,2, BootROM MBA61.0099.B21, 2 processors, Intel Core i7, 1.7 GHz, 8 GB, SMC 2.13f15
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, -
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: HackRF One
Thunderbolt Bus: MacBook Air, Apple Inc., 23.6
```

hb9ari, 2016-03-11:
Le 11.03.2016 08:30, Evan Krall a écrit :
> I was playing around with CubicSDR on my Macbook Air / HackRF One,
> looking at the broadcast FM bands; after a few minutes, it crashed.
> Here's the crash report:
>
> |Process: CubicSDR [88533] Path: [...]

cjcliffe, 2016-03-11:
@EvanKrall looks like a possible race in the DemodulatorThread; I'll take a peek and add an extra mutex lock around the buffer read/write to be sure.

cjcliffe, 2016-04-06:
@EvanKrall give the 0.1.27 release at https://github.com/cjcliffe/CubicSDR/releases a try and let me know if that's any more stable for you.

EvanKrall, 2016-04-07:
Hasn't crashed yet after a few minutes of playing with it.

cjcliffe, 2016-04-08:
@EvanKrall thanks; I'll mark this resolved for now but please feel free to report any new incidents.

closed 2016-04-08 by cjcliffe