cjcliffe/CubicSDR #311: Feature request: digital lab spectrum analyzer

Dantali0n, 2016-03-12:
Hi,

I had this idea that within the digital lab we could have a spectrum analyzer type feature where it would sweep across a frequency range and draw a graph. 

Would be even more awesome if you could zero the graph after taking a measurement of the noise.

Dantali0n, 2016-03-12:
Their is already this program that kinda does that for the rtl-sdr but I found that it didn't work that well but it could help get the idea across: http://www.rtl-sdr.com/new-rtl-sdr-panoramic-spectrum-analyzer/

cjcliffe, 2016-03-15:
@Dantali0n I think I have an issue in for this; if you'd like to expand upon the specific idea it's over at #170 -- I think it might cover this use case.