cjcliffe/CubicSDR #314: upper left (fine tuning?) window goes blank

tedder, 2016-03-17:
[image]

As you can see, even when I have a signal, the upper left window goes black. If I restart it'll come back. I can't figure out what causes it. I can still fine tune it, but the signal isn't there.

This is OSX, 0.1.20-alpha.

cjcliffe, 2016-03-18:
@tedder can you give the build at https://github.com/cjcliffe/CubicSDR/releases/tag/v0.1.25-alpha a try and see if that helps at all? I'd be interested in knowing if this was fixed recently or is still an issue.

Thanks!

tedder, 2016-03-27:
Sorry for the delay. So far I haven't reproduced it on 0.1.25-alpha.

cjcliffe, 2016-04-01:
Sounds good to me; thanks for the report.

closed 2016-04-01 by cjcliffe