cjcliffe/CubicSDR #315: 0.1.26-darwin.dmg - 10.11 only??

hvisage, 2016-03-17:
Hi there,

 Downloaded CubicSDR-0.1.26-Darwin.dmg, and opening it on 10.10.5, it has this circle with a line through it, and running it it complains that it's for 10.11+ only while I have 10.10.5.

 Is that a "bug" as I didn't see the release notes limiting that?
(Or is it the Developer ID signing that went missing? )

Toontje, 2016-03-17:
Confirmed to NOT work on 10.10.5 no matter how you set the security
settings.

dc1rdb, 2016-03-17:
Crashing on 10.9.5:
```
imac:~ rduering$ /Applications/CubicSDR_1.26.app/Contents/MacOS/CubicSDR
Loading:: configuration file '/Users/rduering/Library/Application Support/CubicSDR/config.xml'
Unhandled DataElement toString for type: 0
Loaded PPM for device '' at 0ppm
Loaded offset for device '' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device '192.168.178.32:1001' at 0ppm
Loaded offset for device '192.168.178.32:1001' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Apple Inc.: Built-in Input' at 0ppm
Loaded offset for device 'Apple Inc.: Built-in Input' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Apple Inc.: Built-in Microphone' at 0ppm
Loaded offset for device 'Apple Inc.: Built-in Microphone' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Generic RTL2832U OEM :: 00000001' at 0ppm
Loaded offset for device 'Generic RTL2832U OEM :: 00000001' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Generic RTL2832U OEM :: 00000153' at 0ppm
Loaded offset for device 'Generic RTL2832U OEM :: 00000153' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Generic RTL2832U OEM :: 0000815' at 0ppm
Loaded offset for device 'Generic RTL2832U OEM :: 0000815' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'HackRF One' at 0ppm
Loaded offset for device 'HackRF One' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'SDRPlay RSP' at 0ppm
Loaded offset for device 'SDRPlay RSP' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Terratec NOXON DAB/DAB+ USB dongle (rev 1) :: 0815' at 0ppm
Loaded offset for device 'Terratec NOXON DAB/DAB+ USB dongle (rev 1) :: 0815' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Terratec NOXON DAB/DAB+ USB dongle (rev 2) :: 00000001' at 0ppm
Loaded offset for device 'Terratec NOXON DAB/DAB+ USB dongle (rev 2) :: 00000001' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'ma++ ingalls for Cycling '74: Soundflower (2ch)' at 0ppm
Loaded offset for device 'ma++ ingalls for Cycling '74: Soundflower (2ch)' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'ma++ ingalls for Cycling '74: Soundflower (64ch)' at 0ppm
Loaded offset for device 'ma++ ingalls for Cycling '74: Soundflower (64ch)' at 0Hz
SDR post-processing thread started..
Spectrum visual data thread started.
Spectrum visual data thread started.
FFT visual data thread started.

Audio Device #0 Apple Inc.: Built-in Microphone
	Default Output? No
	Default Input? Yes
	Input channels: 2
	Output channels: 0
	Duplex channels: 0
	Native formats:
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		44100hz
		48000hz
		96000hz

Audio Device #1 Apple Inc.: Built-in Input
	Default Output? No
	Default Input? No
	Input channels: 2
	Output channels: 0
	Duplex channels: 0
	Native formats:
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
[... 56 lines cut ...]
	Install root: /usr/local
	Loading modules... 
Checking local system SoapySDR modules..Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libHackRFSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libRedPitaya.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libairspySupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libaudioSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libbladeRFSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libremoteSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//librfspaceSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//librtlsdrSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libsdrPlaySupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR_1.26.app/Contents/MacOS/modules//libuhdSupport.so..
Mac OS; Clang version 7.0.2 (clang-700.1.81); Boost_105900; UHD_003.009.002-MacPorts-Release

	Available factories...airspy, audio, bladerf, hackrf, null, redpitaya, remote, rfspace, rtlsdr, sdrplay, uhd
dyld: lazy symbol binding failed: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Applications/CubicSDR_1.26.app/Contents/MacOS/../MacOS/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Applications/CubicSDR_1.26.app/Contents/MacOS/../MacOS/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

Trace/BPT trap: 5
```

cjcliffe, 2016-03-17:
I somehow figured that would happen after I upgraded to El Capitan part of why I was holding off...  I'll go back and figure out why my min-version isn't 10.10 as I had it configured -- must be using the wrong CMake variables.

cjcliffe, 2016-03-17:
Yup, looks like I'll have to compile some of the MacPorts dependencies myself for 10.10; looks like some of the libs are built for 10.11 only.

cjcliffe, 2016-03-17:
@hvisage @Toontje @dc1rdb well I think that was easier than I thought but I've probably spoken too soon :)

Check out the latest OSX10.9 suffixed DMG build I've added at https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment -- I removed any MacPorts based dependencies (including libusb) and re-built them manually with 10.9 as the base target; and I didn't get any linker mismatch errors so let's hope it works :) :pray:

cjcliffe, 2016-03-17:
Noticing the 10.9 build is really slow here compared to 10.11; going to investigate and try re-building everything with base target 10.10 as it was before.

cjcliffe, 2016-03-17:
Hmm, slowness isn't 10.9 related -- still a problem in 10.10 build; going to do some profiling and make sure all the dependency build parameters are optimal; it's possible something built debug by default.

cjcliffe, 2016-03-17:
Ok, interesting.. liquid-dsp channelizer is far faster using internal fft instead of compiling against fftw3...  Going to spin up some more test builds..

cjcliffe, 2016-03-18:
CubicSDR-0.1.26-Darwin-OSX10.10.dmg has been added to the releases which appears to be performing better.  Going to try rebuilding 10.9 with the same settings as well.

cjcliffe, 2016-03-18:
The updated 10.9 build is up and working well here; let me know how it goes for everyone -- I think I'll target 10.9 unless there's any issues.

Toontje, 2016-03-18:
10.10 version working fine here. Did you notice the difference in size between the Darwin one and the 10.9 and 10.10 versions?
[image]

cjcliffe, 2016-03-18:
@Toontje 10.11 only one includes UHD, HackRF, BladeRF and SoapyOsmo modules as well; the additional test builds only have a few modules.

dc1rdb, 2016-03-18:
10.9 version working fine here, too.

cjcliffe, 2016-03-18:
All sounds good; going to target 10.9 for releases as I can't find any remarkable differences between it and the 10.10 build.

I'll have a new build ready soon with all of the SDR Modules available; now that I know what I'm targeting I'll put in the extra effort to get the remaining bits built for 10.9 support :+1:

cjcliffe, 2016-03-19:
I've updated the https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment builds page for OSX which now includes all the Soapy SDR support modules and is built for 10.9 minimum.

I'll leave the original test builds here too in case they're needed for reference or comparison:

[CubicSDR-0.1.26-Darwin-OSX10.9.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/180595/CubicSDR-0.1.26-Darwin-OSX10.9.dmg.zip)
[CubicSDR-0.1.26-Darwin-OSX10.10.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/180594/CubicSDR-0.1.26-Darwin-OSX10.10.dmg.zip)

closed 2016-03-22 by cjcliffe