cjcliffe/CubicSDR #316: Idea: no-waterfall mode/branch

jacobseptember, 2016-03-18:
I don't know about other people, but I came to CubicSDR looking for a way—any way—to easily and simply play local FM radio on my Mac using an RTL dongle.
The waterfall is interesting to see, but it uses a fair amount of CPU resources, even with the most recent optimizations. I find myself usually using CubicSDR in the background anyway, as I do other things with my computer.

I was curious—how easy would it be to implement a "simple mode" that just allows for tuning and playing radio, without the fancy waterfalls? If it would be relatively simple I could even give it a shot myself, if I had a hint as to where to start.

cjcliffe, 2016-03-18:
@jacobseptember at some point I'd like to make an equivalent set of SoapySDR based rtl_fm, rtl_sdr, rtl_adsb etc. type tools for the command line using CubicSDR classes and modems which would be a good way to keep it very low overhead for this use case.  Add some basic keyboard or hamlib control support and keep the audio output functionality for piping it over virtual audio as well.

Other than that a working idea for the Lua implementation is to recode the GUI layout and interface code in Lua / wxLua templates which would make it easy to hack on a new interface without recompiling.

But for a more immediate solution I'd be cool with creating a "minimal" mode toggle that strips down the UI to a few elements and reduces the window size (possibly just main spectrum and tuning elements?).  As a quick hint this would basically mean storing all of the UI elements that are created in AppFrame initialization and creating layout functions to enable/disable and disconnect/reconnect the appropriate elements and visual processors for each configuration.  Note that in most cases when you hide a UI element the data to it will stop flowing implicitly since it blocks the visual queue but the waterfall is a special case with it's own threads that can keep buffering in the background until they hit a hard-coded memory limit.

If you wanted to just hide some elements and play with the layouts in the AppFrame init to get started and screenshot something that works for you or just mock it up in an image editor that'd be a good start for discussion on this issue too.

Cheers,

jacobseptember, 2016-03-20:
Cool! I'll try to get everything working with the build & dependencies on one of my machines and see how it goes.