cjcliffe/CubicSDR #317: What happened to Hamlib support?

Toontje, 2016-03-22:
It's been a while since i last compiles CubicSDR from scratch. I did today and noticed that hamlib support is gone. Was this removed or am i doing something wrong?

cjcliffe, 2016-03-22:
@Toontje probably just need to add -DUSE_HAMLIB=1 in the CubicSDR cmake step; it was on by default on the hamlib testing branch we used before.

cjcliffe, 2016-03-22:
Added a build note to the wiki for OSX and Linux on enabling hamlib during cmake command.

Toontje, 2016-03-23:
No, that was not it. I recently updated from Yosemite to El Capitan and my hamlib, installed using MacPorts, was not valid anymore somehow. Reinstalled all my ports including hamlib and all is working again.
You can close this issue. My fault.

closed 2016-03-23 by cjcliffe

cjcliffe, 2016-04-28:
@Toontje latest build up at https://github.com/cjcliffe/CubicSDR/releases now includes bundled hamlib support -- I'd be interested to see if that's working properly for you.  Any issues you find just add a comment to #255 where I'm working on doing the same for Windows and Linux.

Cheers!

Toontje, 2016-04-30:
@cjcliffe I will test as soon as i can. Currently i don't have much time for radio due to work and personal issues i have to take care of. This will take at least a month or more to complete. I would love to help you, but i just don't have the possibilities right now. Sorry.