cjcliffe/CubicSDR #318: Serial port drop-down for Rig Control port

Toontje, 2016-03-23:
Every time i boot my PC my serial port name changes, probably because i am using two Prolific adapters. One port always has the same name, the second port changes whenever i unplug and re-plug it or when i boot my PC.
So every time i want to use Rig Control i need to look up the serial port in my Terminal.

Is it possible to create a serial port drop-down similar to the below enumerating the available serial ports?

[image]

cjcliffe, 2016-03-29:
I use something like this in the Arduino environment regularly; I think I can reproduce it without much trouble.

cjcliffe, 2016-06-14:
I've added this to the list for #373 -- going to move all this stuff into a dialog.

closed 2016-06-14 by cjcliffe