cjcliffe/CubicSDR #319: Zoom presets

Toontje, 2016-03-27:
Can we have configurable zoom presets?
The use case is that i want to be able to see the full HAM band but only this HAM band. For example, i want my waterfall to show from 14.000 - 14.350 MHz or 7.000 - 7.200 MHz. If these presets would be configurable, that would make things really awesome.

cjcliffe, 2016-03-29:
Would a simple range input work?  I.e. I could make it so that if you entered "14-14.350" into the pop-up frequency dialog it would interpret that as a range and set the zoom and center frequency appropriately.  

Eventually I'm going to have that dialog show a list of auto-completes, history, and bookmarks, etc. which might be enough to facilitate "presets" by being able to bookmark the range.

Toontje, 2016-03-30:
Yep! For now that would be ok.

cjcliffe, 2016-08-16:
@Toontje this one too ;)

closed 2016-08-16 by cjcliffe