cjcliffe/CubicSDR #320: Compile error on Ubuntu 14.04

frspin, 2016-04-01:
New version don't compile on Ubuntu 14.04

After a make clean and a cmake .. command this is error message on console:

[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from /home/spin/Scaricati/ham/CubicSDR/src/visual/ScopeCanvas.h:13:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:10,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/InteractiveCanvas.h:11:47: error: ‘wxGLAttributes’ does not name a type
     InteractiveCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                               ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/InteractiveCanvas.h:11:63: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     InteractiveCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                               ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:10:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/ScopeCanvas.h:17:41: error: ‘wxGLAttributes’ does not name a type
     ScopeCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                         ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/ScopeCanvas.h:17:57: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     ScopeCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                         ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:11:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/SpectrumCanvas.h:16:44: error: ‘wxGLAttributes’ does not name a type
     SpectrumCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                            ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/SpectrumCanvas.h:16:60: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     SpectrumCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                            ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:12:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/WaterfallCanvas.h:21:45: error: ‘wxGLAttributes’ does not name a type
     WaterfallCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                             ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/WaterfallCanvas.h:21:61: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     WaterfallCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                             ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:13:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/MeterCanvas.h:17:41: error: ‘wxGLAttributes’ does not name a type
     MeterCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                         ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/MeterCanvas.h:17:57: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     MeterCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                         ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:14:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/TuningCanvas.h:20:42: error: ‘wxGLAttributes’ does not name a type
     TuningCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                          ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/TuningCanvas.h:20:58: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     TuningCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                          ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:15:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/ModeSelectorCanvas.h:27:48: error: ‘wxGLAttributes’ does not name a type
     ModeSelectorCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/ModeSelectorCanvas.h:27:64: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     ModeSelectorCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                                ^
In file included from /home/spin/Scaricati/ham/CubicSDR/src/AppFrame.h:16:0,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.h:23,
                 from /home/spin/Scaricati/ham/CubicSDR/src/CubicSDR.cpp:14:
/home/spin/Scaricati/ham/CubicSDR/src/visual/GainCanvas.h:30:40: error: ‘wxGLAttributes’ does not name a type
     GainCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                        ^
/home/spin/Scaricati/ham/CubicSDR/src/visual/GainCanvas.h:30:56: error: ISO C++ forbids declaration of ‘dispAttrs’ with no type [-fpermissive]
     GainCanvas(wxWindow *parent, const wxGLAttributes& dispAttrs);
                                                        ^
make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Errore 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Errore 2
make: *** [all] Errore 2

Any tip?

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2016-04-01:
@frspin check out the LInux build guide; wxWidgets 3.1.0 is now required so you'll need to re-do that step.

frspin, 2016-04-02:
Il 02/04/2016 01:34, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> check out the LInux build guide;
> wxWidgets 3.1.0 is now required so you'll need to re-do that step.
> 

Ubuntu 14.04 have 3.0.0 version
Ubuntu 15.10 have 3.0.2+dfsg-1.2 and Ubuntu 16.04 will have 3.0.2+dfsg-1.3

Can WxWidgets 3.0.x and 3.1.x live on same system?
Have you tested WxWidgets 3.1.0 version on Ubuntu?

Regards

Franco Spinelli
IW2DHW

cjcliffe, 2016-04-03:
@frspin there's no specific distribution requirements since CubicSDR requires wxWidgets to be statically compiled (i.e. don't use the system packages); if you follow the source build for wxWidgets 3.1.0 you'll be able to get it working.   

I tested the wx 3.1.0 linux build that's on the releases page in Ubuntu 15 without much trouble.

frspin, 2016-04-04:
Il 03/04/2016 02:39, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> there's no specific distribution
> requirements since CubicSDR requires wxWidgets to be statically compiled
> (i.e. don't use the system packages); if you follow the [...]

OK
Compiled wxWidgets with --disable-shared, installed in /usr/local and
now CubicSDR compile OK

Thank you

Franco Spinelli
IW2DHW

closed 2016-04-06 by cjcliffe