cjcliffe/CubicSDR #321: Question: Two frequencies selected?

theforcedk, 2016-04-02:
Hi there

Thanks a lot for working on this great piece of software. I've been using gqrx for a while now; but this is slowly replacing it. Sorry to see gqrx go, but I have a better experience on OS X using this.

I do not understand however, why I see two frequencies "selected" - if you look at the bottom of the screenshot below, you'll see one on the left with the "AM" notation and one on the right with a "USB" notation. This happens here and there, and I've not been able to figure out how to reproduce it, it just shows up. Maybe intentional, hoping you can let me know why this happens and how I should interpret it?

The signal on the right (USB) is the one I actively selected myself.

[image]

cjcliffe, 2016-04-03:
@theforcedk it's likely you've just created multiple modems without realizing it -- i.e. if you create one then drag it off the screen and create another then drag the first one back into view you'll see they're two separate instances.  

In most cases clicking moves the active modem; holding SHIFT (or ALT+SHIFT to drag range) will create a new one instead.  You'll see the active modem turn RED if you're about to move it and pushing SHIFT will turn the indicator GREEN to show you you're creating a new one.

There's no limit to how many modems CubicSDR supports and you never have to delete them you can leave them around and go back to them later if you like.   You can also now use the 'Solo' mode which will limit the audio output to the selected modem only; this way you can set up a bunch of modems and just hit TAB/SHIFT-TAB to toggle listening between them.

Pressing TAB or SHIFT-TAB is a simple way for you to toggle between them; if you see one like this you didn't mean to create then just hit TAB to get to it quickly then tap 'D' to delete.  Zooming in with the mouse wheel or arrow keys on the waterfall will make it easier to hover and press 'D' to delete if you prefer.

Take a look at some of the help tips that show in the status bar or popups (if you haven't disabled) and you'll find a variety of keyboard functions and combinations that are available.  Sometimes the help tip will change to indicate additional actions -- such as pressing SHIFT over the waterfall, or ALT over the frequency bar.

Hope that helps!

theforcedk, 2016-04-03:
@cjcliffe Thanks a lot, Charles. So I can think of "modems" as a type of bookmarks (a modem = demodulator). Would be great to have a list of the modems available somewhere, but I believe I read that in an enhancement issue earlier.

Thanks again for the thorough explanation, and the hotkeys - takes a bit of practice to remember them all, but love the concept of a mouse free environment (vi all the way).

closed 2016-04-03 by theforcedk

ankostis, 2020-12-01:
> in most cases clicking moves the active modem; ...

This succinct description above should go into the manual, as an UX overview.  The current manual is like a "reference" of the UI.

Dantali0n, 2020-12-02:
> > in most cases clicking moves the active modem; ...
> 
> This succinct description above should go into the manual, as an UX overview. The current manual is like a "reference" of the UI.

The manual is open source as well, https://github.com/cjcliffe/CubicSDR-Manual. If you think the description should be part of the manual you can isolate the text and add it in a pull request.