cjcliffe/CubicSDR #322: Error message "Install SoapyRemote module to add remote servers."

TIMBA3E, 2016-04-02:
I'm trying to connect to a SoapyRemote server on my local network. With version v0.1.20 I had no issues connecting, but with v0.1.27 I get the message "Install SoapyRemote module to add remote servers." I'm not too sure what is causing this issue. 

To try and connect I click Add, then select SoapyRemote from the drop down and put the address in the box below. 

Do I need to add any files to the modules folder?

cjcliffe, 2016-04-03:
Hey @TIMBA3E -- what OS is this for?  It's possible I broke the remote module in one of the bundled versions.  I'd be happy to fix it up and let you know when it's updated.

TIMBA3E, 2016-04-03:
Hi Charles, I'm using Windows 10 64bit and the CubicSDR-0.1.27-win64.exe installer. I've also tried v0.1.26 and received the same message also.  

Thanks for you help.

cjcliffe, 2016-04-06:
@TIMBA3E please give it another try -- 0.1.27 windows installers have been updated; my build script went awry and a few of the modules were missed / not built properly last time around.

closed 2016-04-06 by cjcliffe