cjcliffe/CubicSDR #323: Getting a seg fault & symbol lookup error

vsboost, 2016-04-03:
Hi Charles,
Just tried to build the latest and get the below.
-----------------------------------------------------------------
root@kali-sp2:~/Downloads/CubicSDR/build/x64# ./CubicSDR 
Loading:: configuration file '/root/.CubicSDR/config.xml'
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Generic RTL2832U OEM :: 00000001' at 0ppm
Loaded offset for device 'Generic RTL2832U OEM :: 00000001' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'HackRF One' at 0ppm
Loaded offset for device 'HackRF One' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'PulseAudio' at 0ppm
Loaded offset for device 'PulseAudio' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'SDRPlay RSP' at 0ppm
Loaded offset for device 'SDRPlay RSP' at 0Hz
SDR post-processing thread started..
Spectrum visual data thread started.
Spectrum visual data thread started.
FFT visual data thread started.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.5.0-gb16b6d6b
	ABI Version: v0.5-1
	Install root: /usr/local
	Loading modules... 
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libHackRFSupport.so)
  hackrf failed ABI check: Library ABI=0.5-1, Module ABI=0.4-2
	Available factories...null, rtlsdr, sdrplay
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono24.fnt', parsed 167 characters.
Found Rafael Micro R820T tuner
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 0
Found Rafael Micro R820T tuner
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0
Segmentation fault
root@kali-sp2:~/Downloads/CubicSDR/build/x64# 
---------------------------------------------------------------------------------------
Also tried the latest binary and get

Loading:: configuration file '/root/.CubicSDR/config.xml'
Unhandled DataElement toString for type: 0
Loaded PPM for device 'Generic RTL2832U OEM :: 00000001' at 0ppm
Loaded offset for device 'Generic RTL2832U OEM :: 00000001' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'HackRF One' at 0ppm
Loaded offset for device 'HackRF One' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'PulseAudio' at 0ppm
Loaded offset for device 'PulseAudio' at 0Hz
Unhandled DataElement toString for type: 0
Loaded PPM for device 'SDRPlay RSP' at 0ppm
Loaded offset for device 'SDRPlay RSP' at 0Hz
SDR post-processing thread started..
Spectrum visual data thread started.
Spectrum visual data thread started.
./CubicSDR: symbol lookup error: ./CubicSDR: undefined symbol: gdk_mir_display_get_type

Not sure what i have done but thought i should let your know.

Cheers

vsboost, 2016-04-03:
Thought i would attach some more info if it helps.
------------------
root@kali-sp2:~/Downloads/CubicSDR/build# cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config
-- The C compiler identification is GNU 5.3.1
-- The CXX compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
64 bit compiler detected
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found LIQUID: /usr/local/lib/libliquid.so  
-- Found wxWidgets: -L/root/Develop/wxWidgets-staticlib/lib;-pthread;;;/root/Develop/wxWidgets-staticlib/lib/libwx_gtk3u_gl-3.1.a;-lGL;-lGLU;/root/Develop/wxWidgets-staticlib/lib/libwx_gtk3u-3.1.a;-lgtk-3;-lgdk-3;-lpangocairo-1.0;-lpango-1.0;-latk-1.0;-lcairo-gobject;-lcairo;-lgdk_pixbuf-2.0;-lgio-2.0;-lgobject-2.0;-lgthread-2.0;-pthread;-lglib-2.0;-lX11;-lXxf86vm;-lSM;-lgtk-3;-lgdk-3;-lpangocairo-1.0;-lpango-1.0;-latk-1.0;-lcairo-gobject;-lcairo;-lgdk_pixbuf-2.0;-lgio-2.0;-lgobject-2.0;-lglib-2.0;-lpng;-lz;-lwxregexu-3.1;-lwxtiff-3.1;-lwxjpeg-3.1;-lz;-ldl;-lm (found version "3.1.0") 
-- Copying /root/Downloads/CubicSDR/font/*.fnt to directory /root/Downloads/CubicSDR/build/x64
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono16.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono32.fnt
-- Copying /root/Downloads/CubicSDR/font/*.png to directory /root/Downloads/CubicSDR/build/x64
-- Configuring file vera_sans_mono12_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono32_0.png
-- Copying /root/Downloads/CubicSDR/icon/CubicSDR.ico to directory /root/Downloads/CubicSDR/build/x64
-- Configuring file CubicSDR.ico
-- Configuring done
-- Generating done
-- Build files have been written to: /root/Downloads/CubicSDR/build
root@kali-sp2:~/Downloads/CubicSDR/build# make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
[  4%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o
[  5%] Building CXX object CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o
[  7%] Building CXX object CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o
[  8%] Building CXX object CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o
[ 10%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o
[ 11%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o
[ 12%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o
[ 14%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o
[ 15%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o
[ 17%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o
[ 18%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o
[ 20%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o
[ 21%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/Modem.cpp.o
[ 22%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemAnalog.cpp.o
[ 24%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o
[ 25%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o
[ 27%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemDSB.cpp.o
[ 28%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFM.cpp.o
[ 30%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFMStereo.cpp.o
[ 31%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemIQ.cpp.o
[ 32%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemLSB.cpp.o
[ 34%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemUSB.cpp.o
[ 35%] Building CXX object CMakeFiles/CubicSDR.dir/src/audio/AudioThread.cpp.o
[ 37%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/Gradient.cpp.o
[ 38%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/Timer.cpp.o
[ 40%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/MouseTracker.cpp.o
[... 100 lines cut ...]
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/root/Downloads/CubicSDR/build/x64/vera_sans_mono24.fnt', parsed 167 characters.
mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
[... cut ...]
SDR enumerator done.
FFT visual data thread done.

(CubicSDR:2954): GLib-GObject-WARNING **: invalid unclassed pointer in cast to 'GtkWidget'

(CubicSDR:2954): Gtk-CRITICAL **: gtk_widget_set_size_request: assertion 'GTK_IS_WIDGET (widget)' failed
Terminating SDR thread..
Terminating SDR post-processing thread..
SDR post-processing thread done.
Terminating Visual Processor threads..
Spectrum visual data thread done.
Spectrum visual data thread done.
root@kali-sp2:~/Downloads/CubicSDR/build/x64# 
[image]

vsboost, 2016-04-03:
[image]

If i close that box first i can start and all is working.

cjcliffe, 2016-04-05:
@vsboost I'm seeing that dialog too -- I think it's related to some clearing of the glError state I may have disabled -- I'll fix that up and redo a build.

cjcliffe, 2016-04-08:
@vsboost thanks for the help -- pull latest master and the OpenGL error dialog should be fixed up.

vsboost, 2016-04-08:
Issue is now fixed.
Thanks

closed 2016-04-08 by vsboost