cjcliffe/CubicSDR #324: Error upon running SoapySDRUtil --probe (or --info)

n1zhi, 2016-04-03:
Full error message: [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so) sdrplay failed ABI check: Library ABI=0.5-1, Module ABI=0.4-2
Error occurred in Mint 17.3 Mate in a VMware VM and Mint 17.3 Cinnamon on bare hardware (4 core, 32GB RAM)
Both installs were fully updated before attempting this build.
"sudo make install" was run twice in each build as that's how I understood the build steps.
I'm quite willing for this to be an error on my part.

guruofquality, 2016-04-03:
@cjcliffe If it makes things easier, both SoapySDR and SoapyRemote have a "maint" branch which is only fixes since the last release. The master branch is accumulating new API calls and ABI changes which may make it hard to work with pre-built binaries. I have to break things somewhere :-)

guruofquality, 2016-04-03:
@n1zhi I wasn’t totally sure what was a source build or what was binary. If you were manually building the SoapySDR and SoapySDRPlay from source, it looks like libsdrPlaySupport.so was built against the last release of SoapySDR, but SoapySDR was built from the master branch. I suggest re-building SoapySDRPlay after SoapySDR, or when in doubt, delete the installed files in /usr/local, and make clean before re-installing SoapySDRPlay

n1zhi, 2016-04-03:
Hee! I'm familiar with having to break things. That was part of my job 
in IT for 18 years.
I've not yet had to use the Soapy "maint" branch as each build installed 
correctly, with any errors being locally resolved (end-user errors).

I appreciate the response and the information about the maintenance branch.

Ron Cook n1zhi

n1zhi, 2016-04-03:
Thanks again, Josh.
Shall do!

Ron Cook

n1zhi, 2016-04-03:
I reverted my dev system (bare hardware) to Mint 17.2 Cinnamon and, on the
first build received the same error.
As per your suggestions in the previous email, I removed the
libsdrPlaySupport.so, ran make clean, and rebuilt the module.
No errors appeared.
I'd consider this specific issue closed.
Thank you, again, for the assistance.
Ron

closed 2016-04-06 by cjcliffe