cjcliffe/CubicSDR #325: Crash on OS X

theforcedk, 2016-04-03:
Not much information to go by. I had CubicSDR running in the background and was messing around in gnuradio-companion (not running it tho), tabbed into CubicSDR again and hit the "file" menu and it crashed.

Here is the terminal output from start of command until it crashed: http://pastebin.com/2Vpd9iAJ
Here is the OS X crash repport (stacktrace): http://pastebin.com/p5Q7SMsa

theforcedk, 2016-04-04:
Here is another for reference: http://pastebin.com/35JLXH4T

Symptoms: Switching frequencies and gain settings took ~3-6 seconds each time on a cold boot on the HackRF. Tabbing out of the window and navigating to the file menu again, to stop the device in order to try and resolve the slowness (which I have not experienced earlier), crashed.

cjcliffe, 2016-04-04:
@theforcedk thinking I may have pulled a bad HackRF update from master branch for these builds and possibly some experimental Soapy stuff; going to try this here soon and move it back to a more recent stable set of libs.

cjcliffe, 2016-04-04:
@theforcedk there was a bug in my build script that broke some libs which may have been falling back to system; I've also switched to the 'maint' branch of SoapySDR and SoapyRemote and built the latest GitHub libhackrf.   

I've updated the .DMG on the release page at https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.27-alpha which you can grab now.

I've been running it here on HackRF without any crashes or the above mentioned symptoms so far -- hopefully that does the trick.

theforcedk, 2016-04-05:
@cjcliffe Thanks, I'll give it a shot once I get home from the office :). I should mention that I compiled everything from source, all dependencies expect wyPython (via. Homebrew).

theforcedk, 2016-04-05:
@cjcliffe https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.27-alpha crashes right away, probably because I have another SoapSDR lib installed:

<pre>
Checking local system SoapySDR modules..[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libHackRFSupport.so)
  hackrf failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/librtlsdrSupport.so)
  rtlsdr failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
...
Reporting enumeration complete.
SDR enumerator done.
[INFO] Opening HackRF device instance {bias_tx=false, device=HackRF One, driver=hackrf, part_id=a000cb3c006d4f67, serial=0000000000000000406464c823414e4b, version=git-3f4d1a4}...
Bus error: 10
</pre>

Stacktrace:

<pre>
Process:               CubicSDR [19885]
Path:                  /Applications/CubicSDR.app/Contents/MacOS/./CubicSDR
Identifier:            com.cubicproductions.cubicsdr
Version:               0.1.27 (0.1.27-alpha)
Code Type:             X86-64 (Native)
Parent Process:        bash [19819]
Responsible:           Terminal [19817]
User ID:               501

Date/Time:             2016-04-05 16:47:45.718 +0200
OS Version:            Mac OS X 10.11.4 (15E65)
Report Version:        11
Anonymous UUID:        89167778-1716-EC0C-FF57-D715202F7E20

Sleep/Wake UUID:       A4D1F4A1-DAEE-49F9-8A1D-200223E2105E

Time Awake Since Boot: 410000 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007fff9f6af3f8

VM Regions Near 0x7fff9f6af3f8:
    __TEXT                 00007fff9f688000-00007fff9f6a5000 [  116K] r-x/rwx SM=COW  /usr/lib/system/libsystem_malloc.dylib
--> __TEXT                 00007fff9f6a5000-00007fff9f800000 [ 1388K] r-x/rwx SM=COW  /usr/lib/libobjc.A.dylib
    Submap                 00007fff9f800000-00007fffc0000000 [520.0M] r--/rwx SM=PRV  process-only VM submap

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.cubicproductions.cubicsdr 	0x000000010b778a74 std::__1::__vector_base<SoapySDR::ArgInfo, std::__1::allocator<SoapySDR::ArgInfo> >::~__vector_base() + 84
1   com.cubicproductions.cubicsdr 	0x000000010b7e13a3 SDRDevicesDialog::refreshDeviceProperties() + 5603
2   com.cubicproductions.cubicsdr 	0x000000010b7e175e SDRDevicesDialog::OnSelectionChanged(wxTreeEvent&) + 14
3   com.cubicproductions.cubicsdr 	0x000000010b904fe1 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 289
4   com.cubicproductions.cubicsdr 	0x000000010b904ceb wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
5   com.cubicproductions.cubicsdr 	0x000000010b904c07 wxEvtHandler::ProcessEvent(wxEvent&) + 183
6   com.cubicproductions.cubicsdr 	0x000000010ba8f903 wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) + 35
7   com.cubicproductions.cubicsdr 	0x000000010baa4224 wxGenericTreeCtrl::DoSelectItem(wxTreeItemId const&, bool, bool) + 948
8   com.cubicproductions.cubicsdr 	0x000000010ba9da02 wxGenericTreeCtrl::OnMouse(wxMouseEvent&) + 3906
9   com.cubicproductions.cubicsdr 	0x000000010b9035bf wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 431
10  com.cubicproductions.cubicsdr 	0x000000010b904d0a wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
11  com.cubicproductions.cubicsdr 	0x000000010b904c07 wxEvtHandler::ProcessEvent(wxEvent&) + 183
12  com.cubicproductions.cubicsdr 	0x000000010ba8f903 wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) + 35
13  com.cubicproductions.cubicsdr 	0x000000010b905260 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 16
14  com.cubicproductions.cubicsdr 	0x000000010b9a5893 wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 83
15  com.cubicproductions.cubicsdr 	0x000000010b9a1909 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 169
16  com.apple.AppKit              	0x00007fff9c32f469 -[NSWindow _handleMouseDownEvent:isDelayedEvent:] + 6322
17  com.apple.AppKit              	0x00007fff9c33044d -[NSWindow _reallySendEvent:isDelayedEvent:] + 212
18  com.apple.AppKit              	0x00007fff9bd6f63d -[NSWindow sendEvent:] + 517
19  com.cubicproductions.cubicsdr 	0x000000010b9921a5 -[wxNSWindow sendEvent:] + 117
20  com.apple.AppKit              	0x00007fff9bcefb3c -[NSApplication sendEvent:] + 2540
[... 473 lines cut ...]
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       101.4M       47 
__TEXT                           150.1M      257 
__UNICODE                          552K        2 
mapped file                       49.2M       16 
shared memory                     16.4M       10 
===========                     =======  ======= 
TOTAL                            694.2M      836 
TOTAL, minus reserved VM space   689.7M      836 

Model: MacBookPro11,4, BootROM MBP114.0172.B09, 4 processors, Intel Core i7, 2,2 GHz, 16 GB, SMC 2.29f24
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.4f4 17685, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: HackRF One
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
</pre>

cjcliffe, 2016-04-05:
@theforcedk I think my logic of attempting to use the local modules first might be flawed in this case; can you launch it from the command line with:

``` 
/Applications/CubicSDR.app/Contents/MacOS/CubicSDR --bundled
```

And see if there's any difference?  That will prefer the bundled modules over the locally installed.

theforcedk, 2016-04-05:
@cjcliffe Not crashing now; but still get the red message regarding the failed "ABI check" (whatever that is). Seems to still use some SoapySDR modules from /usr/local/lib/. Log below.

Might be wise to always prioritize local bundled modules, because SoapySDR could be used for other things not related to this project (?) - perhaps just mention it in a log file that it's actually using local modules.

<pre>
Checking system SoapySDR modules..[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libHackRFSupport.so)
  hackrf failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/librtlsdrSupport.so)
  rtlsdr failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
	Available factories...airspy, audio, bladerf, hackrf, null, redpitaya, remote, rfspace, rtlsdr, uhd
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono24.fnt', parsed 167 characters.
  default_input = True
  default_output = False
  device_id = 0
  driver = audio
  label = Apple Inc.: Built-in Microphone
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  device = HackRF One
  driver = hackrf
  part_id = a000cb3c006d4f67
  serial = 0000000000000000406464c823414e4b
  version = git-3f4d1a4
Make device 1
[INFO] Opening HackRF device instance {device=HackRF One, driver=hackrf, part_id=a000cb3c006d4f67, serial=0000000000000000406464c823414e4b, version=git-3f4d1a4}...
  clock source=internal
  part id=a000cb3c006d4f67
  serial=0000000000000000406464c823414e4b
  version=git-3f4d1a4

Reporting enumeration complete.
SDR enumerator done.
[INFO] Opening HackRF device instance {bias_tx=false, device=HackRF One, driver=hackrf, part_id=a000cb3c006d4f67, serial=0000000000000000406464c823414e4b, version=git-3f4d1a4}...
Set sample rate: 2000000
Set PPM: 0
Set offset: 0
SDR thread starting.
device init()
Stream MTU: 131072
Calculated optimal 4 channel element count of 66668
Calculated optimal 4 channel element count of 33336
starting readLoop()
</pre>

cjcliffe, 2016-04-05:
@theforcedk Ok -- I'll have a fixed DMG up shortly that should fix remaining errors; just fixing some issues with the windows builds now.

cjcliffe, 2016-04-06:
@theforcedk thanks for your help -- 0.1.27 release DMG has been updated; it will now use bundled modules only unless a user-provided module path is set.

closed 2016-04-06 by cjcliffe