cjcliffe/CubicSDR #326: BadValue (integer parameter out of range for operation)

bdn76, 2016-04-05:
SDR post-processing thread started..
Spectrum visual data thread started.
Spectrum visual data thread started.
The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 229 error_code 2 request_code 154 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

commit 3c51f1a

Linux dima 4.4.0-1-amd64 #1 SMP Debian 4.4.6-1 (2016-03-17) x86_64 GNU/Linux
stretch/sid
wxWidgets commit 7465237

bdn76, 2016-04-05:
broken at f4b7bae commit, 855a6f7 work fine.

cjcliffe, 2016-04-07:
@bdn76 thanks for the pinpoint; I might just back out that attrib list change (my best guess) for now as it's mostly going to be for future OpenGL updates and adds nothing.

cjcliffe, 2016-04-08:
@bdn76 if you pull the latest update I've reverted the wxGLAttribute stuff -- I'll save that experiment for the OpenGL updates.   It appears to have fixed things here and I've updated the Linux AppImageKit bundle as well.

Thanks!

jdeitch, 2016-04-08:
@cjcliffe Just pulled and compiled latest commit and can confirm 1.27 is working on Debian Jessie after this commit.  Previously I was having the exact same error as bdn76.  He beat me to posting it by about 1 hour.

bdn76, 2016-04-08:
At 9ac3104 work fine. Thank you.

closed 2016-04-08 by bdn76