cjcliffe/CubicSDR #329: install_name_tool error (OS X)

theforcedk, 2016-04-07:
I get these errors when running cpack, after cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/usr/local/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 on OS X

The app bundle is registered tho and working, so not a big deal, but I'm not sure why it happens.

<pre>
Alderaan:~/Development/Source/CubicSDR/build cpack
CPack: Create package using DragNDrop
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open input file: /Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib for writing (Permission denied)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't lseek to offset: 0 in file: /Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib for writing (Bad file descriptor)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't write new headers in file: /Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib (Bad file descriptor)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't close written on input file: /Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib (Bad file descriptor)
CMake Error at /usr/local/Cellar/cmake/3.5.1/share/cmake/Modules/BundleUtilities.cmake:789 (message):
  Command failed:

   'install_name_tool' '-id' '@executable_path/../MacOS/libfftw3f.3.dylib' '/Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib'
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.5.1/share/cmake/Modules/BundleUtilities.cmake:850 (fixup_bundle_item)
  /Users/yoda/Development/Source/CubicSDR/build/cmake_install.cmake:34 (fixup_bundle)
  

CPack Error: Error when generating package: CubicSDR</pre>

<pre>
Alderaan:~/Development/Source/CubicSDR/build otool -L /Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib
/Users/yoda/Development/Source/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib:
	/usr/local/opt/fftw/lib/libfftw3f.3.dylib (compatibility version 8.0.0, current version 8.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1225.1.1)
</pre>

cjcliffe, 2016-04-07:
@theforcedk it looks like the files copied from brew might have some sort of sticky permission bit set?  Not sure why it would have copied a file you can't write; but ```sudo rm -rf x64``` in the build folder should at least clear that.   Similarly you might be able to make it build as root using ```sudo cpack``` if just clearing that folder wasn't enough.

On a side note the instructions in the Wiki now suggest disabling fftw in liquid-dsp in favour of internal FFT; the performance improvement was about 10-20% here on average.  If you follow the latest liquid-dsp build info there's a configure parameter to disable it -- CubicSDR now makes FFTW optional by default itself as well.

cjcliffe, 2016-04-07:
@theforcedk optionally you can remove "-DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1" from the CMake command and instead run the CubicSDR binary from the x64/ folder.  There's also a ```make install``` and ```make uninstall``` target that will install it to /usr/local/ so you can just run ```CubicSDR```.

theforcedk, 2016-04-08:
@cjcliffe 

Not sure, homebrew/the recipe sometimes sets weird permissions.

<pre>
Alderaan:~ stat /usr/local/opt/fftw/lib/libfftw3f.3.dylib
16777220 5923257 -r--r--r-- 1 yoda admin 0 1928428 "Apr  8 07:51:23 2016" "Apr  7 21:04:28 2016" "Apr  7 21:04:28 2016" "Apr  7 21:02:18 2016" 4096 3768 0 /usr/local/opt/fftw/lib/libfftw3f.3.dylib
</pre>

Rebuilt liquid-dsp with enable-fftoverride parameter.

When running make, ld throws a warning. /opt/local I believe is where macports puts it's thing, but I do not have this installed, as using homebrew.

<pre>
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
[100%] Linking CXX executable x64/CubicSDR
ld: warning: directory not found for option '-L/opt/local/lib'
[100%] Built target CubicSDR
</pre>

Thanks, I would rather run it from my terminal than /Applications.

Everything seems to run more smooth after a rebuild mentioning what you wrote; how do I check if it's using the internal liquid-dsp library?

theforcedk, 2016-04-08:
@cjcliffe Still appears to be using /usr/local/opt/libfftw. I can remove it tho.

<pre>
23414/0x507e90:  stat64("/System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO\0", 0x7FFF5330F768, 0x2)		 = 0 0
23414/0x507e90:  stat64("/usr/lib/libobjc.A.dylib\0", 0x7FFF5330F768, 0x2)		 = 0 0
23414/0x507e90:  stat64("/usr/local/opt/fftw/lib/libfftw3f.3.dylib\0", 0x7FFF5330F678, 0x2)		 = 0 0
23414/0x507e90:  open("/usr/local/opt/fftw/lib/libfftw3f.3.dylib\0", 0x0, 0x0)		 = 3 0
</pre>

cjcliffe, 2016-04-08:
@theforcedk I just use ```otool -l <bin>``` for checking dependencies; not sure why it would still be included -- I've definitely left it out of the more recent builds without issue -- perhaps a clean build would fix it up?

theforcedk, 2016-04-09:
@cjcliffe Just fetched a clean repo.

Might it be this:

<pre>
Alderaan:~/tmp/CubicSDR grep -rn 'pt/local/lib' .
./cmake/Modules/Findhamlib.cmake:37:		/opt/local/lib
./cmake/Modules/Findhamlib.cmake:38:		/opt/local/lib/hamlib
./cmake/Modules/Findhamlib.cmake:52:		/opt/local/lib
./cmake/Modules/Findhamlib.cmake:53:		/opt/local/lib/hamlib
./CMakeLists.txt:264:    link_directories(/opt/local/lib)
</pre>

cjcliffe, 2016-04-10:
@theforcedk that looks like a likely culprit; one of the last items before beta release is to get hamlib bundled properly so I'll fix that up.

theforcedk, 2016-04-10:
Excellent :)

closed 2016-04-10 by theforcedk

adamwight, 2018-04-24:
@theforcedk I'm running into the same error after following the build guide.  IMO, we should reopen this bug until the build works properly.