cjcliffe/CubicSDR #330: Reception on regular AM band (600khz)

morgan2j, 2016-04-11:
I just started using CubicSDR on Friday after purchasing SDRPlay unit.  Local FM station reception is great, but when I go down to regular AM band most of my local station are not there.  I have a second PC which is an old Windows XP notepad which I loaded HDSDR and I find all my local AM band stations. 
I also found that trying to find WWV at 5Mhz and 10Mhz they were not there, but using HDSDR I found both. 

So.  Am I doing something wrong when trying to receive AM 600khz on CubicSDR, or is the code somehow missing the signals from the SDRPlay device.

Thanks, Jim

theforcedk, 2016-04-11:
Not sure about the SDRPlay implementation (https://github.com/pothosware/SoapySDRPlay ?); but with the HackRF implementation I have to set manual gain settings, as auto is really screwy.

morgan2j, 2016-04-11:
Hello,

I did just turn off the auto gain and was able to now see AM broadcast stations peak on the spectrum display.  I’m able to now (sort of) hear the station I was looking for, but it’s so low in the noise that I can’t really understand what they are saying.  Using a regular AM radio it comes in weak, but you can listen to it.  

I will keep playing around with options to see if I can get rid of the white noise and still receive the station.

Obviously I am really new to the SDR game and will keep trying.

Thank you for your quick response.

Jim

cjcliffe, 2016-04-11:
@morgan2j What type of antenna do you have connected? Sounds like you might just need an antenna more sensitive to AM frequencies -- I wasn't able to get much in the way of AM or ham bands until I set up a long wire antenna (just a length of speaker wire wrapped around perimeter of my apartment ceiling) and a 1:9 balun.

cjcliffe, 2016-04-11:
@morgan2j here's my quick and dirty setup I've been using for about a year; it's quite effective at ~100Khz - 1.8Mhz, 3Mhz, 6Mhz, 7Mhz and 9Mhz but above that and areas in-between it's pretty weak:

[image]

morgan2j, 2016-04-12:
I’m not sure I received what you were trying to send.

My antenna is a temporary longwire, and it’s not very long so I suspected poor reception.  However, I have a second (old) windows PC and just for fun I loaded DHSDR and I can clearly hear my local stations.  When you send your recommended setting I will try, and I will also look at the setting on HDSDR to see if I can find a setting that will help.  

I did improve the reception a lot when I dropped the gain down to about 48, but still hard to understand.  I am starting to think it’s my macbook audio compared to my windows pc.

Thanks, Jim

morgan2j, 2016-04-13:
I spent some more time yesterday evening trying different gain settings.  Reception and clarity seems great except for 630Khz AM.  Once I start going up I see and can listen to many stations.  Up around 5Mkz, 10Mkz, and 15Mkz things work great on AM and side band.  Just for fun I tuned in some aircraft tower communications and worked great.

So, for me listening to 630Khz is not important.  

On a side observation I noticed that if I  lower the gain control way down (about 3/4 from bottom) everything jumps way back up even thought the control is way down.  Above the 3/4 mark it works great.

Jim

closed 2016-04-26 by cjcliffe