cjcliffe/CubicSDR #332: SDRPlay - OSX - Waterfall issue

cetola, 2016-04-14:
The waterfall seems to be displaying data until it gets to the bottom of a sweep, at which point it "resets". I've had one other confirmation of this on the SDRPlay forums:

http://www.sdrplay.com/community/viewtopic.php?f=10&t=678

I've reproduced this issue on multiple versions, including 0.1.27.

cjcliffe, 2016-04-14:
That's really strange.. that texture wraps vertically so there shouldn't be any way for it to "reset" the entire data set is always visible and it just moves the texture pointer by a single pixel..

Is there a version you can't reproduce the issue with?

cetola, 2016-04-14:
So far I've tested 0.1.18, 0.1.26, and 0.1.27. I can test any other versions if you want, just let me know.

Interesting note: if there is a spike in the intensity of a signal, that spike will still be visible, but it will have those vertical lines obscuring the image.

[image]

Hopefully that helps, but let me know if there's any other information I can provide, or any way I can help.

cjcliffe, 2016-04-14:
@cetola I'll try making a test build with some additional wrapping calls and post it here to try shortly.

cjcliffe, 2016-04-14:
@cetola give this build a try; basically all I did was add a call to set the texture wrap flag after texture init and again before each draw -- fairly harmless change if it works: [CubicSDR-0.1.27-Darwin-extrawrap.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/220097/CubicSDR-0.1.27-Darwin-extrawrap.dmg.zip)

cetola, 2016-04-14:
Sadly no change:

[image]

cjcliffe, 2016-04-14:
@cetola here's another attempt: [CubicSDR-0.1.27-Darwin-wrap2.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/220110/CubicSDR-0.1.27-Darwin-wrap2.dmg.zip) -- if this works for some reason your graphics card's driver might be weird :)

cetola, 2016-04-14:
Success!  I clearly have a strange graphics card driver :)

  Model Identifier:	MacBookPro12,1
...
Intel Iris Graphics 6100:

  Chipset Model:	Intel Iris Graphics 6100
  Type:	GPU
  Bus:	Built-In
  VRAM (Dynamic, Max):	1536 MB
  Vendor:	Intel (0x8086)
  Device ID:	0x162b
  Revision ID:	0x0009

Well done!

cjcliffe, 2016-04-14:
@cetola Haha, yeah I wondered if that might be it -- apparently your driver is confused about the X and Y axis of the texture coordinate system with respect to wrapping rules; nothing else made sense so that was a long shot ;)

I'm going to make it a bit less aggressive and post one more build if you could try to confirm that it still works.

cetola, 2016-04-14:
@cjcliffe Of course, happy to help.

cjcliffe, 2016-04-14:
@cetola here goes -- let me know if this one still works: 
[CubicSDR-0.1.27-Darwin-wrap3.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/220125/CubicSDR-0.1.27-Darwin-wrap3.dmg.zip)

cetola, 2016-04-14:
@cjcliffe Looks great; runs smoothly too.  Cheers!

cjcliffe, 2016-04-14:
@cetola cheers! Thanks for the help -- will commit that to master and update the current release build in a few minutes.

closed 2016-04-14 by cjcliffe

mehdideveloper, 2016-04-15:
Thanks. It is solved here too (Mac OS X 10.11.4)