cjcliffe/CubicSDR #333: Airspy not working with 0.1.27

gilbertoverg, 2016-04-15:
With version 0.1.25 all is working fine, but with 0.1.27 all spectrums and waterfall remains blank

[image]

cjcliffe, 2016-04-15:
@gilbertoverg hmm, I compiled my own copy of the Airspy host this time around so I probably did something strange -- I'll post another build in this issue soon for you to try.

cjcliffe, 2016-04-20:
@gilbertoverg here's a build that uses the latest official release version instead of the GitHub master version of the airspy host -- let me know if this works any better: [CubicSDR-0.1.27-Darwin.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/227118/CubicSDR-0.1.27-Darwin.dmg.zip)

gilbertoverg, 2016-04-20:
@cjcliffe test build works perfectly. Thank you!

cjcliffe, 2016-04-21:
@gilbertoverg good to hear; I'll update the 0.1.27 release build with this fix -- cheers!

closed 2016-04-21 by cjcliffe