cjcliffe/CubicSDR #334: Do not put AppImage into a zip file

probonopd, 2016-04-16:
As you might be aware, the AppImage is an ISO file that mounts itself at runtime. It is already compressed, so you gain relatively little by putting it into a zip file. However, the user has to unpack it. This is not necessary if you just provide the AppImage itself for download. 

Possibly you did this so that the file is executable without specific user intervention. Granted, if you provide the AppImage for download rather than a zip then the user has to set the executable bit, but this is easy in most GUIs. Also, if you keep the ".AppImage" extension, then users can find help easily on what to do with this file.

Finally, your AppImage requires at least glibc 2.15 which means that it will not run on distributions older than 2012ish. Is that acceptable for you? Otherwise, you could compile on an older build host distribution which would increase the compatibility of the resulting AppImage.

cjcliffe, 2016-04-16:
@probonopd thanks! AppImageKit is working well so far and I'm looking to improve what I can -- I'm going to investigate everything you've suggested here for the 0.2.0 release build.   

I'll probably ping you here again soon if I have any questions :)

probonopd, 2016-04-16:
@cjcliffe By the way, the AppImage is working great. Running it on Ubuntu 16.04 and listening to some FM radio. 

But I see an error saying

```
(CubicSDR:4003): Gtk-CRITICAL **: gtk_distribute_natural_allocation: assertion 'extra_space >= 0' failed
[ERROR] SoapySDR::loadModule(././/local/lib/SoapySDR/modules/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so: cannot open shared object file: No such file or direct
```

cjcliffe, 2016-04-19:
@probonopd Nothing to worry about; that's the SDRPlay module being noisy about not finding the mirsdr lib since it's not included in the bundle -- it needs to be installed separately since they're only available in binary form on the SDRPlay site.

cjcliffe, 2016-04-29:
@probonopd I've added a new build -- no longer compressed, CubicSDR bin is stripped (saved around 1MB) and I've added the .AppImage extension.

Going to stick with glibc 2.15 as SoapySDRPlay requires it, but I'll re-think making an alternate build if there's demand.

Thanks again for all your hard work and assistance!

closed 2016-04-29 by cjcliffe