cjcliffe/CubicSDR #335: Ask user to unload kernel modules

probonopd, 2016-04-16:
I had to manually run

```
sudo rmmod rtl2832_sdr
sudo rmmod lirc_dev dvb_usb_rtl28xxu dvb_usb_v2
sudo rmmod  ir_sharp_decoder ir_xmp_decoder lirc_dev ir_lirc_codec dvb_usb_rtl28xxu ir_rc5_decoder ir_nec_decoder ir_sony_decoder ir_mce_kbd_decoder ir_jvc_decoder dvb_usb_v2 ir_rc6_decoder ir_sanyo_decoder
```

before launching the AppImage in order for CubicSDR to be able to access the hardware. It would be nice if the GUI would inform the user that he as to do this (and optionally do it for him after entering the root password).

cjcliffe, 2016-04-18:
@probonopd Hmm -- I think the only one I've ever had to rmmod is dvb_usb_rtl28xxu -- this would be specific to the RTL-SDR module but I think it would be possible to throw an error from the SoapySDR Module that CubicSDR could interpret or just make it an rtl-sdr specific check for linux -- I'll have to think about that one.

cjcliffe, 2026-11-14:
Latest version can auto-detach the existing rtl driver making this unnecessary; thanks!

closed 2026-11-14 by cjcliffe