cjcliffe/CubicSDR #336: Bug: preview top-left window freezes when user tap a frequency lower than 256KHz

dmitryelj, 2016-04-17:
Hi guys,
Thanks for the nice program.

I've found a bug using my SDRplay: top-left spectrum preview window freezes when I tap the frequency lower than 256000Hz. 256001 is ok :)
Receiver: SDRplay
Sample rate: 2M
Version: 0.1.27a, Mac OS.

Please check.
Thanks.

morgan2j, 2016-04-18:
Also using 0.1.27 - waterfall in both windows resets when it reaches the bottom.  Hope you have tine to look at this also.  I like Cubic a lot !

cjcliffe, 2016-04-18:
@dmitryelj this sounds familiar; I think I've got an unnecessary restriction in there -- I'll definitely patch that up.

@morgan2j when's the last time you pulled a copy of 0.1.27?  It was updated on the weekend for OSX to address a problem that sounds exactly like that so it *may* be working now -- sorry I've been updating the builds with minor patches as I'm trying to roll up and test all the remaining bug fixes without pushing a bunch of scattered minor releases before the beta :)

morgan2j, 2016-04-19:
I loaded about two weeks ago.  I will download and test this evening.  

Thank you for all your hard work - it’s appreciated.

If I have a suggestion for Cubic, where can I look to make sure someone else hasn’t already suggested?

Thanks again,

Jim

cjcliffe, 2016-04-19:
@morgan2j No problem; thanks for the kind words -- There's a search box near the top of the page at https://github.com/cjcliffe/CubicSDR/issues just empty the input box and search using a keyword or two to see if there's a relevant open or closed issue.

morgan2j, 2016-04-20:
Thank you.

I loaded the latest  0.1.27 update and the waterfall issue is fixed.

Jim

cjcliffe, 2016-06-04:
Had an unnecessary frequency check that was disabling visual data in the lower-most channel when using the zoomed-view.  

Fixed in master and will be included in the 0.2.0-beta-rc1 release shortly.

closed 2016-06-04 by cjcliffe