cjcliffe/CubicSDR #337: Suggestion:  Save Settings

morgan2j, 2016-04-20:
When manually turning off AGC and then adjusting both the LNAT and Tuner sliders to your liking it would be nice if you could manually (or auto) save the setting.  That way each future launch of the application everything would be set up and ready to go.

What I am using today: 
SDRPlay & CubicSDR 0.1.27
Macbook Air

Thanks, Jim

cjcliffe, 2016-04-20:
@morgan2j I think I have an issue that includes saving the gain state; but I'll keep this one open for 0.2.0 feature visibility -- settings are always saved on exit (mostly in case you pick something that crashes) but I'll consider making it manageable once the settings can be saved and loaded safely during runtime.

cjcliffe, 2016-05-12:
AGC State and gains are now saved per-device; even if you turn AGC back on your last gain settings won't be lost when you return.

It's merged to master and will be included in the next binary build.

closed 2016-05-12 by cjcliffe

morgan2j, 2016-05-12:
That’s great, thank you.

Jim