cjcliffe/CubicSDR #340: Run multplice instances of CubicSDR ?

neilwillgettoit, 2016-04-30:
Is it possible to run multiple instances of CubicSDR simultaneously ?

What I'm trying to do: 
Launch 2 instances of CubicSDR to connect to two different devices. First launches and claims the first device with no issue, but when I have OSX launch a second instance and claim the second device, both instances crash out.

theforcedk, 2016-05-01:
@neilwillgettoit Try launching them both from the terminal, and provided the output of the crash and/or the Apple crash report (use PRE (< pre >) tags if pasting). I can launch two instances just fine. RTL stick on one, and HackRF on the other.

neilwillgettoit, 2016-05-01:
Going through my logs it looks like the issue isn't with CubicSDR, but SoapySDRServer segfaulted. I haven't been able to recreate this issue this morning.

closed 2016-05-01 by neilwillgettoit