cjcliffe/CubicSDR #341: Feature Request: CW modem

neilwillgettoit, 2016-05-04:
Hi,

It would be very nice to have a cw modem. Right now the usb/lsb can be used to listen, but when incorporating rig control / hamlib, it ends up requiring a tuning offset to shift the carrier into an audio tone.

TonMachielsen, 2016-05-04:
@cjcliffe Looks like the Digital Lab needs to get a kick in the butt again. Or just leave it out and rely on third party software using audio pipes.

vsboost, 2016-05-04:
I'm in no rush.

Toontje, 2016-05-04:
@cjcliffe Looks like the Digital Lab needs to get a kick in the butt again. Or just leave it out and rely on third party software using audio pipes.

neilwillgettoit, 2016-05-04:
@Toontje I'm not looking for any type of decoder. Just a button for cw on the mode list so I can tune in morse signals with a bit more resolution than a usb/lsb filter, and with the side tone offset so the frequency resolution matches my rig vfo.

neilwillgettoit, 2016-05-04:
For example: 

When listening to a busy band like 40m, you will get multiple stations that fit within the standard 2.7khz ssb filter: 

[image]

Even if I chop the filter down, because it is only reducing the top end of the ssb window, I cannot filter out other stations that are near and below the signal I am trying to focus on: 

[image]

And then there is still the issue with sidetone and how that conflicts with rig control.

cjcliffe, 2016-05-06:
@neilwillgettoit this is definitely something I'd like to try to do as a decoder eventually; @Toontje is on track as I've already kinda tried it with the liquid OOK decoder :)

I'll start with the CW filter mode and see what I can do about feeding it to the OOK decoder for interpretation in the future.

Toontje, 2016-10-16:
Just getting into CW now and i see what @neilwillgettoit means. I +1 the request for a CW modem. Currently what i do is set the filter width to 500 and listen in USB/LSB, but it's not ideal.

krisbee, 2018-09-14:
I guess just a +1 for this feature.  I basically can use this software for everything in conjunction with my ham setup - listen on my rtl-sdr, control the rig (by following modem), transmit, etc except for when I’m on CW where I basically have to abandon listening if Im operating because of the TX offset.  If this modem was present with a 500hz filter as default (and then you can adjust if needed) and a programmable offset (I use 750hz) then Id never actually have to use the receiver in my rig ever again, which is a great thing.

Aplonis, 2019-07-30:
A CW filter, all by itself, with no decoding, adjustable perhaps only for bandwidth, would serve my own needs very completely. Some days I have spent, trying to get other SDR programs working on Linux to no avail. Always they fail in some way or other, none of which I have yet had any luck figuring out. But CubicSDR worked right away, apparently in every way except this one feature, which so far, it totally lacks. What I had been hoping to do was to use SDR on Linux as my ham station receiver. This while still making use of my 30-year-old Icom IC-745 as my HF transmitter. And or my 10m QRP transmitter. Or the 630m transmitter kit which I bought and plan to build. I only ever work CW. Not for contests, but mainly to rag chew. My average QSO durations across all bands these past 30 years is above 24 minutes per QSO. The overall average, with many of those lasting an hour. And at age 64, my hearing is not at all what it once was. The 270 Hz crystal filters in the IC-745 ring very badly, and for that reason, are no longer much use to me. But the SDR transceivers of other op's which I saw last Field Day had CW at 300 Hz filter clear as a bell. I really want that, but can't work a new rig into my budget. My laptops are Linux, save for the single Windows machine which I use for work. Would really like CW via SDR on Linux. And CubicSDR seems like the best bet ... as it just works. Except that I only ever work CW. So, if you please, count this as yet another vote in favor of adding the feature. No decoding needed for me. I have that inside of my head. I just need to hear it more clearly is all. That would be great. Please to keep it on the to-do list. Thanks. 73, de KY8D

n2iw, 2019-08-14:
I need CW mode also, but instead of add a new decoder for CW, why not just use USB decoder like normal RIGs, just adjust a sidetone frequency in the display! 

I'm actually using USB mode (500Hz bandwidth) to receive CW which works just fine! What I really need is to shift the modem frequency base on sidetone. So I can sync with my RIG, and use my HDRPlay as panadapter and  second receiver!

colbyAtCRI, 2021-01-14:
I've written a CW modem. It takes the IQ base band and interpolates to audio sampling rate. This is then frequency shifted by a user selectable "beep" frequency. I have several questions. The current default bandwidth is 500Hz. What would a good default be?

krisbee, 2021-01-14:
On Thu, Jan 14, 2021, at 6:22 PM, Paul Colby wrote:
> 

> I've written a CW modem. It takes the IQ base band and interpolates to audio sampling rate. This is then frequency shifted by a user selectable "beep" frequency. I have several questions. The current default bandwidth is 500Hz. What would a good default be?

> 

> 
> 

I think 500hz is perfect.

colbyAtCRI, 2021-01-15:
So, I've cloned the latest and greatest master, [here](https://github.com/colbyAtCRI/CubicSDR), to which I've added ModemCW. The default frequency offset is 650Hz (user settable) with a default Bandwidth of 500Hz. One may turn auto gain on or off and choose to set the audio gain manually (in dB). The keyboard shortcut (which logically should be 'C') isn't used as it would interfere with the centering feature which also uses 'C'. I would suggest using 'Shift C' for centering and 'C' for selecting ModemCW. I'm open to suggestions.

cjcliffe, 2021-03-20:
@colbyAtCRI merged now; tested here and working well.  Thanks for your contribution :+1: 

I'll think about the keyboard shortcut for it; it might be a good idea to add a modifier key like shift for all the modem-specific shortcuts; especially if the digital set becomes part of the default build.

closed 2021-03-20 by cjcliffe

bbsound, 2024-08-17:
If this is still a place to contact cjcliffe - loving cubicsdr on silicon m3 mac with sdrplay hardware - is there any way to get a dBm scale in the spectrum display ?  I’m looking at wireless FM signals for pro audio and a handy dandy scale to see signal to noise ratio would be super…  many thanks!