cjcliffe/CubicSDR #342: Feature request: Make Peak Hold display also for current demod view

vsonnier, 2016-05-05:
I don't know if that would be really usefull or not, since I'm no specialist: 
Make the Peak Hold option also display  a peak hold on the detailed, current demodulator view which looks more precise and reflecting more the "real" shape of the signal.

cjcliffe, 2016-05-06:
@vsonnier good call; it's already supported by that view I've just never thought to enable it.  Adding this to the enhancements, thanks!

vsonnier, 2016-06-09:
I've implemented it in https://github.com/cjcliffe/CubicSDR/pull/368. It just works.

cjcliffe, 2016-06-09:
looking good thanks!  I've also just added 'P' as the keyboard peak-hold toggle key.

closed 2016-06-09 by cjcliffe

vsonnier, 2016-06-10:
Indeed, P as global shortcut is convenient. Thanks !