cjcliffe/CubicSDR #343: [, ] nudge dosn't work on OSX with non english keyboard layout

effeerre75, 2016-05-10:
It works fine switching to english layout.

cjcliffe, 2016-05-10:
@effeerre75  Hmm, do those keys still produce '[' and ']' when typed in non-english or does it map to something else?

effeerre75, 2016-05-11:
In the Italian layout, for example, they produce "è" and "+". Then "[" and "]" are produced by pressing "alt è" and "alt +"  but dosen't work on CubicSDR. You can easly check it adding a new keyboard layout and then enabling the software keyboard. Let's me know if you need more info!
Thanks!

cjcliffe, 2016-05-19:
@effeerre75 give this build a go and let me know if it works for you:

[CubicSDR-0.1.28-Darwin-keytest.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/271767/CubicSDR-0.1.28-Darwin-keytest.dmg.zip)

Cheers!

effeerre75, 2016-05-19:
Works really fine thanks! It work also without pressing the "alt" key: this is really handy!
Have a nice day!

cjcliffe, 2016-05-19:
good to hear, if there's others like that I have a way to catch them now.

Cheers!

closed 2016-05-19 by cjcliffe