cjcliffe/CubicSDR #345: CubicSDR problems on OS X 10.11.4

mehdideveloper, 2016-05-12:
Hi
I've used CubicSDR successfully in the past, but in the last releases I've got 2 problems:
1- I compiled CubicSDR, SoapySDR and SoapySDRPlay from the source and installed them.
When I connect my SDRPlay, I can find it by running "SoapySDRUtil --find"
Now when I run my compiled CubicSDR (v0.1.28-alpha), it can find the SDRPlay, but it's extremely slow. Clicking on buttons, takes more than 4-5 seconds to show any feedback. When I change the frequency, it freezes again for about 5 seconds, then it works as expected.

2- I downloaded the 0.1.28 binary and ran it. On the  first dialog (SDR Devices), it hangs for about 10 seconds with this message:
"Initializing bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules/libuhdSupport.so"
I moved "libuhdSupport.so" to another folder to see whether that makes any difference, and now it shows the same message with "libsdrPlaySupport.so"
Anyway, after this 10-second delay, it just shows the "Apple Built-In Microphone" as my local device and can't find SDRPlay.

Any ideas what could've gone wrong?
I'm using a MacBook Pro 13" 2015 with OS X 10.11.4

Thanks
Mehdi

mehdideveloper, 2016-05-12:
Update:
I solved the first problem by installing the SDRPlay API 1.8.1 (I had 1.8.0)
The second problem still remains.

cjcliffe, 2016-05-13:
@mehdideveloper yeah the SDRPlay requiring the separate install will occasionally mean it's out of sync with the SDRPlay release binaries but I don't think I can do anything about that unless I can trap that version error somehow.

issue 2 - I've seen happen fairly often; in almost every case it was because I had my SDR device connected through a hub or had some other USB device on the same bus that interfered

Try unplugging any hubs, long extensions or unused USB devices and see if it helps; if not there might be something else going on..  You can run CubicSDR from the command line with:

```
/Applications/CubicSDR.app/Contents/MacOS/CubicSDR
```

which might yield some additional log info in the terminal.

mehdideveloper, 2016-05-14:
OK, here's my test results:
I ran the 0.1.28-alpha which I've downloaded from your website. It shows this error:

```
.....
	Available factories...airspy, audio, bladerf, hackrf, redpitaya, remote, rfspace, rtlsdr, sdrplay
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  remote failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so)
  sdrplay failed ABI check: Library ABI=0.4-2, Module ABI=0.5-1
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono16.fnt', parsed 167 characters.
....
```
It shows my SDRPlay after a 10-second delay.

Now when I run the version that I've compiled (again 0.1.28-alpha), it just shows this warning (and it not slow):

```
.....
mir_sdr_2500_Bypass001Trim: reg13=0x1cd reg14=0x6c014e
[WARNING] mir_sdr version: '1.810' does not equal build version: '1.800'
  mir_sdr_version=1.810000

Reporting enumeration complete.
SDR enumerator done.
Loaded font 'Bitstream Vera Sans Mono
......'
```

So it seems it's related to ABI version difference.

cjcliffe, 2016-05-14:
@mehdideveloper ah yes this isn't what I thought; I see the issue -- this was encountered recently and the next binary build will actually not attempt to load '/usr/local/lib/SoapySDR/modules/' unless explicitly requested at the command line or by environment-variable.  It was a useful default when everyone's modules were in sync with the test builds but not so much when using binary bundles.

Next time the binary builds will just load the bundled modules and should completely avoid the double-loading and delays unless requested in that scenario.

closed 2016-05-16 by cjcliffe