cjcliffe/CubicSDR #346: Multiple frequencies

quantegy, 2016-05-15:
For some reason, and I'm not sure when or why this happens, but when I select a frequency on the waterfall, or in the CVFO/VFO, the software starts adding additional frequencies, and I hear multiple stations at once. I had noticed it when I went to mute the audio, and it wouldn't mute. The spectrum will also show multiple frequencies selected/marked at one time. I'll follow up with a screenshot the next time it happens. My setup is SDRPlay on Mac El Capitan 10.11.4

cjcliffe, 2016-05-16:
That's a feature to be able to listen to multiple frequencies (and choose a separate audio output for each if desired) but only when you're holding SHIFT key or there's no VFO currently selected.

Your SHIFT key might have been stuck, VFO wasn't selected or you created one.  If you check the hover tips you'll see when you hover over the VFO on the waterfall you can press 'D' to delete it.

As a note you'll see when you hover in the waterfall if you're about to "move" a VFO it'll show as red; if you're about to create a *new* VFO it'll appear as green.

Hope that helps!

quantegy, 2016-05-16:
Okay, thanks. I thought it was broken. Great software.

cjcliffe, 2016-05-17:
@quantegy Thanks; sorry I couldn't point you at proper instructions yet but I'm going to have a first round draft of the manual up for 0.2.0-beta release, only about 4 issues left to clear up before then :)

closed 2016-05-17 by cjcliffe