cjcliffe/CubicSDR #347: Menu changes to IF not always sent to SDRplay hardware

ka9q, 2016-05-16:
Hi. I'm running CubicSDR v0.1.20-alpha on a Macbook Pro running 10.11.4, and it appears that SDR front end parameters are not always updated when I change menu settings that I expect would change them.
In particular, changing the IF mode sometimes does not actually do anything. This is most easily seen when using the 1 MHz sampling rate. A strong carrier component shows up outside the receiver bandpass exactly 450 kHz above the center frequency even when "Zero-IF" is selected and "Low IF" is off. Sometimes by resetting the session I'm able to force it into a particular state but the problem recurs.
I know that hardware front ends vary and you're trying to abstract them with device drivers, but it would be nice to have direct control over all receiver hardware settings or at least some information on how commands are mapped to hardware settings. "Sample rate" does the obvious thing but "Low IF" is apparently undocumented. There also appears to be an AGC even when "Automatic Gain" is unchecked; perhaps it's implemented in software?

ka9q, 2016-05-16:
I should add that I've run CubicSDR from the command line so I could observe the debugging messages, and sure enough it often sets (e.g.) the use of a 450 kHz IF even when I've checked Zero IF.

cjcliffe, 2016-05-17:
@ka9q apologies as that's correct at the moment; there's a new version of SoapySDRPlay in the works that should be able to dynamically change the IF and bandwidth; for now you can really only tell it whether or not to use Low-IF when appropriate and not actually enable or change it.

closed 2016-09-02 by cjcliffe