cjcliffe/CubicSDR #348: 0.2.0 Manual Spec

cjcliffe, 2016-05-19:
- [ ] First round, top-level explanation of major features
- [ ] Determine best place and platform for Manual
 - [ ] Simple Maintenance
 - [ ] Community Contribution 
 - [ ] Versioning
- [ ] Annotated videos? Inline screencap GIFs?
- [ ] Standard tools and process for any media
- [ ] Level of detail?

neilwillgettoit, 2016-05-19:
I'm down to help with this however I can. Just a few ideas/questions as well: 

- [ ] specific how to tutorials ? 
- [ ] hamlib/rigctl configurations examples? ie - wjst control / decoding  or vfo tracking on a hardware rig over tcp. 
- [ ] hardware / platform list that works or doesn't work? ie rpi2/3 limits 
- [ ] when and where documentation should shift over to pothosware wiki items and keep consistent over time?

cjcliffe, 2016-05-19:
@neilwillgettoit I'm going to aim for mostly a top-level application usage specific overview for the first pass.

Case-specific stuff like how-to guides will be handy once you can reference steps back to the manual pages instead of having to explain the entire process each time as well.

I'll probably try to reference as much of the pothos guides as I can to fill in some parts of the build guides eventually; I definitely am duplicating some build instructions, etc. on some of my pages that would be better suited to the soapy module wikis.

The manual this spec is for is actually issue #248 so check that out as a rough outline of what I'm planning.

cjcliffe, 2016-05-20:
Just tinkering with http://cubicsdr.readthedocs.io/ at the moment :)

cjcliffe, 2016-05-20:
Looks pretty good; I've set up a repository for it https://github.com/cjcliffe/CubicSDR-Manual which makes it nice and simple to manage contributions.

theforcedk, 2016-05-21:
Very nice Charles, good job. Read the docs should fit right into the workflow that is already applied; easy to contribute (pull requests).

cjcliffe, 2016-06-23:
@neilwillgettoit thanks for your help; I've restructured the Index a bit and have been working on the overview for the Main Application Window; work-in-progress can is at http://cubicsdr.readthedocs.io/en/latest/application-window.html

closed 2016-09-01 by cjcliffe