cjcliffe/CubicSDR #349: Question: CubicSDR as a frontend; output to fifo or similar?

theforcedk, 2016-05-21:
Hi Charles

Just a quick question. Is it possible (or on a roadmap) to output the stream/bits directly to a file/pipe or similar, so that we can pipe the input into another application for further processing? For example directly into dump1090 using the parameters "-ifile -".

Thanks :)

neilwillgettoit, 2016-05-24:
@theforcedk If you're on linux with pulseaudio you should be able to add to the pulseaudio configuration a module-pipe-sink and specify the file where you want to send the fifo stream. Then while running cubicSDR go to pavucontrol and re-point pulseaudio to use that new sink as an output.

theforcedk, 2016-05-25:
@neilwillgettoit Thanks for the tips :) I'm on OS X tho, but I'll try and see if I can find alternatives.

closed 2016-05-25 by theforcedk

cjcliffe, 2016-05-25:
@theforcedk #207 will address this eventually; it's something I'd like to get working myself.