cjcliffe/CubicSDR #350: After leaving fullscreen, window is stuck below top menubar

quantegy, 2016-05-25:
After leaving fullscreen mode on El Capitan 10.11.5, the CubicSDR top bar is not viewable. It sits just below the Mac tool bar at the top, and I cannot move the window from that spot.

[image]

quantegy, 2016-05-25:
Had to hide the mac menu bar, and drag the window down.

neilwillgettoit, 2016-05-25:
@quantegy which version of cubicSDR are you running? And do you have the steps to reproduce?

quantegy, 2016-05-25:
@neilwillgettoit Sorry about that, I didn't realize you can't see the version in the titlebar =)

It's version 0.1.18-alpha

Steps:

1. Went into fullscreen mode (clicked the little green dot in top left part of the window on Mac)
2. Cmd-Q to exit CubicSDR
3. Restarted CubicSDR

It did not reopen in fullscreen mode, but the window filled the entire screen, and title bar was sitting below the Mac menu bar.

cjcliffe, 2016-05-25:
@quantegy not a solution to this problem but a tip in general on OSX; if you put your mouse on the edge of a window and click and drag parallel to the edge (up/down on the sides or left/right on the bottom) it will let you grab and move the window from *any* edge that's visible.

closed 2016-08-16 by cjcliffe