cjcliffe/CubicSDR #351: Idea for the Future

morgan2j, 2016-05-26:
As an old shortwave DXer I really miss the tuning knob which would allow me to slowly tune across the spectrum without even looking at the radio.

I’ve tried to use the trackpad on my Macbook Air by holding down the ‘right click’ and scrolling across the spectrum.  This is cumbersome to use and difficult to stop exactly on the signal.

How about in a future design enhancement using the  ‘Caps Lock’ to activate the left and right arrow keys to slowly tune across the spectrum.  This would be a one finger operation that could also be speed controlled by a single click or hold down the arrow to move faster.  To turn off you undo to ‘Caps Lock’.

I know you are very busy with big fixes and enhancements that are probably way more important than this idea, but maybe down the road you could consider.  Or canvass other users to see if anyone else might be interested in this feature.

Thanks, 

Jim

neilwillgettoit, 2016-05-26:
Hi @morgan2j try hovering your mouse over the frequency digit you wish to tune by and use two fingers to scroll up and down on your trackpad.

morgan2j, 2016-05-27:
I just had a chance to try your suggestion.  Really easy to use!

Thank you.

Jim

cjcliffe, 2016-05-30:
For an additional tuning mode similar to this you can also click and drag any frequency digit left or right to change it at varying speed; the further you drag the more it accelerates.

cjcliffe, 2016-06-03:
There's demod drag, mousewheel, digit drag and nudge keys ( ']' and '[' ) that can all surf frequencies fairly leisurely; going to close this one up as I think it's already handled by one or more of those.

closed 2016-06-03 by cjcliffe