cjcliffe/CubicSDR #352: font size

imesg, 2016-05-27:
Read a thread about font size which was closed Jan 13. I did not understand how to adjust the fonts but I did move the separators with no effect. I have a dell xps 13 with a 3800x1800 screen. This may not be a CubicSDR issue but the fonts are too small to read with the naked eye unless I get about 5 inches away from the screen. I need a magnifier to read the screen. I can read the frequencies below the 'Audio Visuals'. The menu bar is OK also. I struggle with this in other programs also. I am trying to sort the font issue out. If you have any suggestion for adjusting the fonts it may help me.

Thanks

cjcliffe, 2016-05-27:
Ah yes; I'll leave this open but I haven't had a chance to work with a retina screen; I can attempt to make some assumptions and simulate it here and see how it goes though.

imesg, 2016-05-28:
I guess I should have mentioned I have installed this on gentoo linux. CubicSDR really looks good on this display except for font size which may be a OS scaling issue.
Thanks

cjcliffe, 2016-06-03:
@imesg can I get a screenshot of the window maximized?  I think I can use that to approximate the next font size up that would be needed to make it appear properly at that resolution.

imesg, 2016-06-17:
cjcliffe did you get the screenshot I sent?

vsonnier, 2016-06-17:
I'm working on a way to use bigger menu settable fonts on a branch of my cloned repo. @cjcliffe I can work on it and you'll have a look when it is ready.

cjcliffe, 2016-06-17:
@vsonnier that would be great; even if there was just a simple small/med/large option. 👍

vsonnier, 2016-06-17:
@cjcliffe That would be normal (like now) and médium (1.33x) and large 2x for instance. l'm ok to take this issue, so that you can concentrate on more important matters like doing à SoapyAirspy module. 😃

ElectronicBattle, 2016-06-17:
hello

When I do ./CubicSDR.AppImage

I get a fault:

./CubicSDR.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version 
`CXXABI_1.3.8' not found (required by ./CubicSDR.AppImage)
./CubicSDR.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version 
`GLIBCXX_3.4.20' not found (required by ./CubicSDR.AppImage)
./CubicSDR.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version 
`GLIBCXX_3.4.20' not found (required by ./lib/libSoapySDR.so.0.5-1)

I can see it's SoapySDR related - please can you suggest what I ought to 
do about it?

vsonnier, 2016-06-23:
@ElectronicBattle Maybe you should report the problem into the issue https://github.com/cjcliffe/CubicSDR/issues/363, which seems more appropriate. This one is dedicated to the font size specifically, which is technically applicable to every OS :)

vsonnier, 2016-06-23:
@cjcliffe My branch https://github.com/vsonnier/CubicSDR/tree/settable_font_sizes_with_menu is coming along nicely, although I have an idea left to turn GLFont.drawString() to someting more universal, independent of the actual font used underneath, with more control and less hacks in "user" code.

Anyway, all this should be ready for you to review before the end of the week.

vsonnier, 2016-06-23:
@cjcliffe pull request sent !

cjcliffe, 2016-06-23:
@vsonnier thanks, checking it now!

@ElectronicBattle I think I'll to try some AppImageKit alternatives (FlatPak, FPM) as well and put them up for everyone to compare, I'll post it over at #363 when something is ready to test.

cjcliffe, 2016-06-29:
Works for me for now; can open new issues if we need specific font adjustments.

closed 2016-06-29 by cjcliffe