cjcliffe/CubicSDR #353: Mouse Wheel ranges for meter canvases

cjcliffe, 2016-05-27:
- The meter canvases now support mouse wheel but haven't been calibrated to their ranges
- The manual gain sliders don't support mouse wheel yet.

vsonnier, 2016-05-30:
@cjcliffe  I tried the  https://github.com/cjcliffe/CubicSDR/compare/mousewheel_tweak branch (ce1cd2700724c77312dd42f61bf7f361c9730f38) and so far the pros and cons compared to master are:
-  At startup the focus is OK
-  The oscillo part no longer react to mousewheel, just the waterfall when the mouse is hovering. I think it is more natural to enable also the oscillo to zoom, like in master.
-  Still no mousewheel for gain, but maybe it is not there yet:)
-  Waterfall and squelch meters are faster, but not enough, and all others have become really slow, I guess they have the same scale applyed on them all.

I tried the folowing modification of MeterCanvas.cpp:
``````cpp
void MeterCanvas::OnMouseWheelMoved(wxMouseEvent& event) {
    InteractiveCanvas::OnMouseWheelMoved(event);
	float movement = 3.0 * (float)event.GetWheelRotation() ;
``````

And it looked much more responsive that way: I counted 28 wheel-clicks now to cover a whole range, but I guess it may be much platform DPI dependent...

cjcliffe, 2016-05-30:
@vsonnier thanks for the report -- I'll give your suggestion a go it looks simpler than mine -- and I haven't added support to gain yet but that's next on the list.

cjcliffe, 2016-05-30:
@vsonnier I've pushed a few updates to mousewheel_tweak -- let me know how that works for you.

vsonnier, 2016-05-31:
Tried the mousewheel_tweak branch, everything is fine now on my side, with the Manual gain still to have  mousewheel support. Thank you !

cjcliffe, 2016-05-31:
@vsonnier hah; I forgot to actually add focus to the gain so that it works in Win7 -- I've pushed a fix for that now; likely going to merge to master shortly so I can keep working on the session loading bug (which I've reproduced here now in Win10 from your xml)

closed 2016-05-31 by cjcliffe

vsonnier, 2016-06-01:
@cjcliffe Gain is working now, except the meter is escaping to +-infinity:)

That actually fixes it in GainCanvas.cpp:

``````cpp
void GainCanvas::OnMouseWheelMoved(wxMouseEvent& event) {
    InteractiveCanvas::OnMouseWheelMoved(event);
  ...
        gInfo->current = gInfo->current + ((movement / 100.0) * ((gInfo->high - gInfo->low) / 100.0));

        //BEGIN Clamp to prevent the meter to escape :)
        if (gInfo->current > gInfo->high) {
            gInfo->current = gInfo->high;
        }
        if (gInfo->current < gInfo->low) {
            gInfo->current = gInfo->low;
        }
        //END Clamp to prevent the meter to escape :)

        gInfo->changed = true;
        
        float levelVal = float(gInfo->current-gInfo->low)/float(gInfo->high-gInfo->low);
        gInfo->levelPanel.setSize(1.0, levelVal);
        gInfo->levelPanel.setPosition(0.0, levelVal-1.0);

        gInfo->valuePanel.setText(std::to_string(int(gInfo->current)));
    }
``````