cjcliffe/CubicSDR #354: Session loading weirdness, demodulators don't load in the right place

cjcliffe, 2016-05-27:
From @vsonnier -- #309 

@cjcliffe I have a (somewhat reproducible) use case for duplicated, stack upon themselves demodulators showing up at loading of an XML session file:

- launch CubicSDR,
- choose RTL-SDR,
- uncheck auto gain, set 25dB instead,
- load DMR_430Mhz_ref.xml:

`````xml
<?xml version="1.0" ?>
<cubicsdr_session>
    <header>
        <version>0.1.27-alpha</version>
        <center_freq>430263415</center_freq>
        <sample_rate>2560000</sample_rate>
    </header>
    <demodulators>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430575946</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.75</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430287741</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>11660</bandwidth>
            <frequency>430263415</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>0</muted>
            <active>1</active>
        </demodulator>
        <demodulator>
            <bandwidth>15000</bandwidth>
            <frequency>430100589</frequency>
            <type>FM</type>
            <squelch_level>-29.7744</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Default Device</output_device>
            <gain>0.125</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430050956</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
        <demodulator>
            <bandwidth>12500</bandwidth>
            <frequency>430175376</frequency>
            <type>FM</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>CABLE Input (VB-Audio Virtual Cable)</output_device>
            <gain>0.740385</gain>
            <muted>1</muted>
        </demodulator>
    </demodulators>
</cubicsdr_session>
````

Now most of the time, a demodulator get reloaded with duplication over itsef, here the 430.263Mhz. Also the Delta Lock (V) looks activated for this demodulator:

![dmr_430mhz_duplicated_demodulator](https://cloud.githubusercontent.com/assets/5152742/14992107/b2568d8e-1164-11e6-8320-212a2e10eca5.jpg)

- If you reload the XML file, the demodulators looks OK:
![dmr_430mhz_reload_ok](https://cloud.githubusercontent.com/assets/5152742/14992159/f9902c6e-1164-11e6-862a-bb3b7e1a4392.jpg)
 
Here is the global config.xml file of CubicSDR:
``````xml
<?xml version="1.0" ?>
<cubicsdr_config>
    <window>
        <x>-9</x>
        <y>-9</y>
        <w>1920</w>
        <h>968</h>
        <max>1</max>
        <tips>0</tips>
        <low_perf_mode>0</low_perf_mode>
        <theme>5</theme>
        <snap>1</snap>
        <center_freq>430263415</center_freq>
        <waterfall_lps>51</waterfall_lps>
        <spectrum_avg>0.549296</spectrum_avg>
    </window>
    <devices>
        <device>
            <id>CABLE Output (VB-Audio Virtual Cable)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Default Device</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Generic RTL2832U OEM :: 00000001</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts>
                <buffers>15</buffers>
                <bufflen>16384</bufflen>
            </streamOpts>
            <settings>
                <direct_samp>0</direct_samp>
                <iq_swap>false</iq_swap>
                <offset_tune>false</offset_tune>
            </settings>
            <rig_ifs />
        </device>
        <device>
            <id>Microphone (Realtek High Definition Audio)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
        <device>
            <id>Mixage stéréo (Realtek High Definition Audio)</id>
            <name></name>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts />
            <settings />
            <rig_ifs />
        </device>
    </devices>
</cubicsdr_config>
``````

vsonnier, 2016-05-29:
@cjcliffe I think it's definitly a problem of initialization. I recompiled the master in a Visual C++ 2015 Update 2 x64 binary. 
By chance I remarked that activating the [/sdl] (https://msdn.microsoft.com/en-us/library/jj161081.aspx) option 
make the problem actually desappear. 
I think the point to consider is:  
````
Performs class member initialization. 
Automatically initializes all class members to zero on object instantiation (before the constructor runs).
This helps prevent the use of uninitialized data associated with class members that the constructor does not explicitly initialize.
````

cjcliffe, 2016-05-31:
@vsonnier hmm, this makes sense -- I tried fixing it as a race condition in Windows 7; but I've just reproduced the issue on Win10 and OSX; so it's definitely universal.

I'll take a look at class init and see what I can find; this might fix a few random issues with new demodulator instances...

cjcliffe, 2016-06-01:
@vsonnier with our latest fixes and cleanups I'd have expected this issue to at least improve; but now the failure actually seems more consistent.. I'm starting to think that there's actually something wrong with the XML loader variable init -- going to break down the loading process and target exactly where they're being lost.

cjcliffe, 2016-06-02:
Hmm, starting to see a bit of a pattern after cleaning up the log now..  missing modems end up around 64.743 in this run..

```
Garbage collected demodulator instance 430.175
Garbage collected demodulator instance 430.051
Garbage collected demodulator instance 430.101
Garbage collected demodulator instance 430.263
Garbage collected demodulator instance 430.288
Garbage collected demodulator instance 430.576
	Added demodulator at frequency 430575946 type FM
	Added demodulator at frequency 430287741 type FM
	Added demodulator at frequency 430263415 type FM
	Added demodulator at frequency 430100589 type FM
	Added demodulator at frequency 430050956 type FM
	Added demodulator at frequency 430175376 type FM
Garbage collected demodulator instance 430.175
Garbage collected demodulator instance 430.051
Garbage collected demodulator instance 430.101
Garbage collected demodulator instance 430.263
Garbage collected demodulator instance 430.288
Garbage collected demodulator instance 430.576
	Added demodulator at frequency 430575946 type FM
	Added demodulator at frequency 430287741 type FM
	Added demodulator at frequency 430263415 type FM
	Added demodulator at frequency 430100589 type FM
	Added demodulator at frequency 430050956 type FM
	Added demodulator at frequency 430175376 type FM
Garbage collected demodulator instance 430.175
Garbage collected demodulator instance 430.051
Garbage collected demodulator instance 430.101
Garbage collected demodulator instance 430.263
Garbage collected demodulator instance 430.288
Garbage collected demodulator instance 64.743
	Added demodulator at frequency 430575946 type FM
	Added demodulator at frequency 430287741 type FM
	Added demodulator at frequency 430263415 type FM
	Added demodulator at frequency 430100589 type FM
	Added demodulator at frequency 430050956 type FM
	Added demodulator at frequency 430175376 type FM
Garbage collected demodulator instance 430.175
Garbage collected demodulator instance 430.051
Garbage collected demodulator instance 430.101
Garbage collected demodulator instance 64.743
Garbage collected demodulator instance 430.288
Garbage collected demodulator instance 430.576
	Added demodulator at frequency 430575946 type FM
	Added demodulator at frequency 430287741 type FM
	Added demodulator at frequency 430263415 type FM
	Added demodulator at frequency 430100589 type FM
	Added demodulator at frequency 430050956 type FM
	Added demodulator at frequency 430175376 type FM
```

cjcliffe, 2016-06-03:
We have a winner.. 
https://github.com/cjcliffe/CubicSDR/pull/358/commits/5a6d97e4805c709242fcebbe0d5a703583b80658 :) 

Fixed in #358 

deltaLockOfs atomic_bool was never initialized; not sure why that didn't seem to show up in XCode or MSVC code analysis..

closed 2016-06-03 by cjcliffe